FlyGoat/RyzenAdj #146: Version 0.8.1 gives error in Task managet

jumpingjackflash5, 2021-05-06:


Falcosc, 2021-05-06:
Could you share some more details?

jumpingjackflash5, 2021-05-06:
¨
[image]

Here in the results of a task was seen the error code 0xc0000005 I wrote above.
With 0.8.0 there is written Task finished succesfully (my screen is in Czech language).

jumpingjackflash5, 2021-05-06:
My task has got simple command line, I do not use the scripts that are included.

[image]

Falcosc, 2021-05-06:
@jumpingjackflash5 please try to get the error message by sending the command output somewhere

`c:\ryzenadj\ryzenadj.exe --tctl-temp=70 >> c:\ryzenadj\output.log`

jumpingjackflash5, 2021-05-06:
On command line it is OK. It givees that error just as a result in the task scheduler.

Falcosc, 2021-05-06:
> On command line it is OK. It givees that error just as a result in the task scheduler.

please use `>> path\logfile` in your task scheduler and post the log when you get the error again.

jumpingjackflash5, 2021-05-06:
I tried but that files is empty, even if I run it from batch file. When run from command line everything seems to be OK 
[image]

Falcosc, 2021-05-06:
`c:\ryzenadj\ryzenadj.exe --tctl-temp=70 >> c:\ryzenadj\output.log`

Will output only to output.log, try it in CMD and if it is working, you can use it in scheduler.

jumpingjackflash5, 2021-05-06:
No. Version 0.8.1 does not output to log regardless of from where it is run. Only screen output from command line is OK.
Reverting to 0.8.0 which works completely OK.

Falcosc, 2021-05-06:
Does 0.8.0 output to `c:\ryzenadj\output.log` ?

jumpingjackflash5, 2021-05-06:
Yes, perfectly

Falcosc, 2021-05-06:
Thank you, found the error.

jumpingjackflash5, 2021-05-06:
Ok, happy to hear that!

Falcosc, 2021-05-06:
@jumpingjackflash5 have a look at the 0.8.2 build:
https://github.com/FlyGoat/RyzenAdj/actions/runs/817695928

closed 2021-05-06 by Falcosc

jumpingjackflash5, 2021-05-06:
Runs succesfully now!