FlyGoat/RyzenAdj #149: PCI Bus is not writeable, check secure boot || Ubuntu

ajitjadhav28, 2021-05-10:
root@elise:/home/ajit# ryzenadj --power-saving
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
PCI Bus is not writeable, check secure boot
Unable to get MP1 SMU Obj
Unable to init ryzenadj

OS Details: Ubuntu 20.04.2 LTS
Kernel: Ubuntu 20.04.2 LTS
CPU Details: AMD Ryzen 5 4600H with Radeon Graphics

Any Suggestions ?

ajitjadhav28, 2021-05-10:
Thanks for reply :smile: 
Disabling secure boot helped. Now I can use it. :man_dancing: 
On windows 10 it was working even with secure boot enabled. Is it linux specific problem?
I am closing this Issue.

Thanks again :100:

closed 2021-05-10 by ajitjadhav28

Falcosc, 2021-05-11:
Yes, Linux does disable PCI access if it detects user wants a hardened system by selecting secure boot.