FlyGoat/RyzenAdj #150: [feature request] Extend --info with current status of power-saving/max-performance

slashbeast, 2021-05-10:
Please consider adding a way to probe what's the current setting, power-saving or max-performance, maybe on --info page. Currently there seems to be no way to get this information.

Falcosc, 2021-05-11:
it does change 3-6 Values known as "Setpoint"

> ```
> --power-saving     Hidden options to improve power efficiency (is set when AC unplugged): 
>                        behavior depends on CPU generation, Device and Manufacture
> --max-performance  Hidden options to improve performance (is set when AC plugged in): 
>                        behavior depends on CPU generation, Device and Manufacture
> ```
> 
> Each time you change the power source your device will automatically set AC or DC mode to tune some unknown values for power usage. We found out that on Zen, Zen+, Zen2 and Zen3 it does change 3-6 Values known as "Setpoint". These are busy thresholds for frequencies and define how fast you the APU does start to boost: Commonly known as boost delay. We don't know what else gets tuned, and we don't know if each device does use these modes to change the boost delay.

We could display 1 of them
It will be either 50 or 95 as thresholds. And we can pair it with the actually calculated busy value.
I would select the CPU usage one because without capability to fine-tune we don't need so many details

I would rename it to 
CPU Boost SetPoint
CPU Busy

Would this fit your usecase?

slashbeast, 2021-05-11:
Hi,

> Would this fit your usecase?

Of course!

I'd love to script it and add it to my Conky as well as dmenu script so I can toggle them around, but I need to be able to probe which is the current value. I can align to absolute any way you would interface it, so feel free to choose what you believe is the best, anything will work for me.

slashbeast, 2021-05-11:
Oh, this is what I've missed. Indeed I can use 0x00FC as the indicator, which has 95 on power-saving and 50 on max-performance modes.

In that case I have everything I need to work with, so this feature would be nice to have, but has rather low priority for me since I can get what I need out of dump table

Thanks mate!

slashbeast, 2021-05-11:
Reports correct value, 95 for power saving and 50 for max performance, as well as fluctuating CCLK BUSY VALUE. ✔

PR #153 (2021-05-11): Add Core Clock Boost setpoint and Core Clock busy value
Resolves #150
commit a3871a22 (2021-05-11): Add Core Clock Boost setpoint and Core Clock busy value
Resolves #150
closed 2021-05-11 by Falcosc