FlyGoat/RyzenAdj #151: tctl-temp reported 0.000 on --info and on HWiNFO

bagusnl, 2021-05-11:
Hi,

I am currently using Ryzen 3 3200U and RyzenAdj 0.8.1. When I do `ryzenadj -i` the tctl-temp limit value is shown correctly at 95.000 but the actual value is 0.000 causing the TCTL Temp reporting on HWiNFO to be 0.000 as well.

```
.\ryzenadj -i
CPU Family: Picasso
SMU BIOS Interface Version: 8
Version: v0.8.1
PM Table Version: 1e0004
```

|       Name       |   Value   |      Paramter      |
|------------------|-----------|--------------------|
| STAPM LIMIT      |    25.000 | stapm-limit        |
| STAPM VALUE      |    10.092 |                    |
| PPT LIMIT FAST   |    21.000 | fast-limit         |
| PPT VALUE FAST   |     7.930 |                    |
| PPT LIMIT SLOW   |    25.000 | slow-limit         |
| PPT VALUE SLOW   |     9.829 |                    |
| StapmTimeConst   |   200.000 | stapm-time         |
| SlowPPTTimeConst |    60.000 | slow-time          |
| PPT LIMIT APU    | -nan(ind) | apu-slow-limit     |
| PPT VALUE APU    | -nan(ind) |                    |
| TDC LIMIT VDD    |    45.000 | vrm-current        |
| TDC VALUE VDD    |     1.809 |                    |
| TDC LIMIT SOC    |    15.000 | vrmsoc-current     |
| TDC VALUE SOC    |     3.307 |                    |
| EDC LIMIT VDD    |    55.000 | vrmmax-current     |
| EDC VALUE VDD    |    37.494 |                    |
| EDC LIMIT SOC    |    20.000 | vrmsocmax-current  |
| EDC VALUE SOC    |     5.233 |                    |
| THM LIMIT CORE   |    95.000 | tctl-temp          |
| THM VALUE CORE   |     0.000 |                    |
| STT LIMIT APU    | -nan(ind) | apu-skin-temp      |
| STT VALUE APU    | -nan(ind) |                    |
| STT LIMIT dGPU   | -nan(ind) | dgpu-skin-temp     |
| STT VALUE dGPU   | -nan(ind) |                    |

Full power table dump provided below

[dump-table.log](https://github.com/FlyGoat/RyzenAdj/files/6456548/dump-table.log)

Falcosc, 2021-05-11:
I will take one of the other values
| Offset |    Data    |   Value   |
|--------|------------|-----------|
| 0x0050 | 0x42BE0000 |    95.000 |
| 0x0054 | 0x00000000 |     0.000 |
| 0x0058 | 0x42BE0000 |    95.000 |
| 0x005C | 0x423CFCBA |    47.247 |
| 0x0060 | 0x42BE0000 |    95.000 |
| 0x0064 | 0x00000000 |     0.000 |
| 0x0068 | 0x42BE0000 |    95.000 |
| 0x006C | 0x423F23EF |    47.785 |
| 0x0070 | 0x42BE0000 |    95.000 |
| 0x0074 | 0x423BABFA |    46.918 |
| 0x0078 | 0x42BE0000 |    95.000 |
| 0x007C | 0x4238BF01 |    46.187 |
| 0x0080 | 0x42BE0000 |    95.000 |
| 0x0084 | 0x423F0E1C |    47.764 |

bagusnl, 2021-05-11:
The issue seems to be something set the value of vrm_current to 0.000 when tctl reported by HWiNFO goes to 75 C
```
CPU Family: Picasso
SMU BIOS Interface Version: 8
Version: v0.8.1
PM Table Version: 1e0004
|       Name       |   Value   |      Paramter      |
|------------------|-----------|--------------------|
| STAPM LIMIT      |    20.000 | stapm-limit        |
| STAPM VALUE      |    12.830 |                    |
| PPT LIMIT FAST   |    20.000 | fast-limit         |
| PPT VALUE FAST   |     4.301 |                    |
| PPT LIMIT SLOW   |    25.000 | slow-limit         |
| PPT VALUE SLOW   |    13.336 |                    |
| StapmTimeConst   |   200.000 | stapm-time         |
| SlowPPTTimeConst |    60.000 | slow-time          |
| PPT LIMIT APU    | -nan(ind) | apu-slow-limit     |
| PPT VALUE APU    | -nan(ind) |                    |
| TDC LIMIT VDD    |     0.000 | vrm-current        |
| TDC VALUE VDD    |     1.306 |                    |
| TDC LIMIT SOC    |    15.000 | vrmsoc-current     |
| TDC VALUE SOC    |     1.646 |                    |
| EDC LIMIT VDD    |    55.000 | vrmmax-current     |
| EDC VALUE VDD    |     2.915 |                    |
| EDC LIMIT SOC    |    20.000 | vrmsocmax-current  |
| EDC VALUE SOC    |     2.003 |                    |
| THM LIMIT CORE   |    95.000 | tctl-temp          |
| THM VALUE CORE   |     0.000 |                    |
| STT LIMIT APU    | -nan(ind) | apu-skin-temp      |
| STT VALUE APU    | -nan(ind) |                    |
| STT LIMIT dGPU   | -nan(ind) | dgpu-skin-temp     |
| STT VALUE dGPU   | -nan(ind) |                    |
```

Falcosc, 2021-05-11:
I had Prochot issues on some workloads. I had to go down to 93°C

But your TDC LIMIT VDD goes to 0A discovery is new for me.

Maybe this is the thing which is limiting some people.

Is your Thermal Throttle PROCHOT active during this 0A state?
How many seconds are you limited at 0A? Maybe it makes sense to check if limit is 0A and try to overwrite it.

bagusnl, 2021-05-11:
PROCHOT readings doesn't seems to be active during 0A state, but it just might be because the CPU throttles so hard HWiNFO fails to read the value because the system is basically almost freezed during 0A state. The 0A state seems to active for 10s to 20s

Falcosc, 2021-05-11:
The PowerShell script is tuned to use as few cycles as possible. You can use it to set in a loop.

If your system is setting the 0A only once every X seconds, you may can workaround by overwriting it.

bagusnl, 2021-05-11:
Yeah, I'm currently using the powershell script with inpoutx64.dll deleted due to Anti-Cheat issue, I'm checking the power table from different directory containing the same RyzenAdj version.

I'm not the best on PowerShell, might need help to set it up. Is it on line 33 ? The current string is this `$Script:repeatWaitTimeSeconds = 1` haven't changed anything on the powershell script other than the ones for the config.

Falcosc, 2021-05-11:
No, something else is setting 0A every X secounds

X is unknown, you have to find this out by testing it.

`$Script:repeatWaitTimeSeconds = 1` does only control how often the script does its work

Using the monitorField function is already a good start, but maybe it is too slow, because each time you use it, it will set your whole configuration. Your 9 Adjustment calls are translated to about 50 calls to your CPU power management.

I guess you can improve your results by setting the current limit first, so move it on top of your configuration

bagusnl, 2021-05-11:
Wait, question. Since I deleted inpoutx64.dll is it gonna affect the monitorField function?

bagusnl, 2021-05-11:
I see, at this point I think there is no other option than doing that since its either having a risk of getting hardlocked during game/working due to 0A state, or can't play some games because inpout dll.

I don't see too much CPU usage from either powershell or ryzenadj calls so I think its okay for now. Still kind of upset that the 0A state always happens on 75C regardless of tctl_temps overwrite though, maybe there is a way to disable that?

Falcosc, 2021-05-11:
Be careful with using 2 scripts, this could hit a known issue:
https://github.com/FlyGoat/RyzenAdj/issues/138

One of 1000 calls could fail if both run at the same time, pretty slim odds, but still possible.

bagusnl, 2021-05-11:
Thank you for making me aware of that issue, I might use powershell script for the vrm_current and AATU for everything else for the sake of simplicity and easier to maintain at the moment, so if anything fails to apply I can reapply it fairly easily.

Just kind of a random question though. Is it possible to get something like Easy Anti Cheat to allow this exact software to use the inpout driver by lets say contacting them? Since RyzenAdj (and its derrivatives) is starting to get more popular as more AMD Ryzen laptops making its way into costumers

Falcosc, 2021-05-11:
@bagusnl please have a look on https://github.com/FlyGoat/RyzenAdj/actions/runs/832192125 and check if temp reading does work.

bagusnl, 2021-05-11:
> @bagusnl please have a look on https://github.com/FlyGoat/RyzenAdj/actions/runs/832192125 and check if temp reading does work.

Temp reading looks okay on that artifact
```
 .\ryzenadj -i
CPU Family: Picasso
SMU BIOS Interface Version: 8
Version: v0.8.2
PM Table Version: 1e0004
|       Name       |   Value   |      Paramter      |
|------------------|-----------|--------------------|
| STAPM LIMIT      |    20.000 | stapm-limit        |
| STAPM VALUE      |     9.223 |                    |
| PPT LIMIT FAST   |    20.000 | fast-limit         |
| PPT VALUE FAST   |    11.155 |                    |
| PPT LIMIT SLOW   |    25.000 | slow-limit         |
| PPT VALUE SLOW   |     9.176 |                    |
| StapmTimeConst   |   200.000 | stapm-time         |
| SlowPPTTimeConst |    60.000 | slow-time          |
| PPT LIMIT APU    | -nan(ind) | apu-slow-limit     |
| PPT VALUE APU    | -nan(ind) |                    |
| TDC LIMIT VDD    |    45.000 | vrm-current        |
| TDC VALUE VDD    |     1.902 |                    |
| TDC LIMIT SOC    |    15.000 | vrmsoc-current     |
| TDC VALUE SOC    |     1.087 |                    |
| EDC LIMIT VDD    |    55.000 | vrmmax-current     |
| EDC VALUE VDD    |    37.219 |                    |
| EDC LIMIT SOC    |    20.000 | vrmsocmax-current  |
| EDC VALUE SOC    |     2.001 |                    |
| THM LIMIT CORE   |    95.000 | tctl-temp          |
| THM VALUE CORE   |    47.158 |                    |
| STT LIMIT APU    | -nan(ind) | apu-skin-temp      |
| STT VALUE APU    | -nan(ind) |                    |
| STT LIMIT dGPU   | -nan(ind) | dgpu-skin-temp     |
| STT VALUE dGPU   | -nan(ind) |                    |
```

closed 2021-05-11 by Falcosc

Falcosc, 2021-05-12:
@bagusnl please interrupt your work on the workarround script. Somebody with a similar issue shared the solution.

You need to install nitrosense
https://global-download.acer.com/GDFiles/Application/Nitro%20Sense/Nitro%20Sense_Acer_3.01.3016_W10x64_A.zip?acerid=637399032918120303&Step1=NOTEBOOK&Step2=NITRO&Step3=NITRO%20AN515-43&OS=ALL&LC=en&BC=ACER&SC=PA_6

After installation you should have access to a power plan named "Balanced (Acer Optimized)"

This plan does fix the Limit issue according to 2 other users.

Please give us feedback because I am not sure if the 2 other users didn't confirm the 0A limit, so I am not sure if this does solve your isse.

Falcosc, 2021-05-12:
Somebody was so kind to share the plan for you

[Acer Powerplan from Nitrosense.zip](https://github.com/FlyGoat/RyzenAdj/files/6465677/Acer.Powerplan.from.Nitrosense.zip)

https://winaero.com/export-import-power-plan-windows-10/

This user did share it: https://discord.com/channels/772105072720871435/772118974640685076/841991393123893248

bagusnl, 2021-05-12:
That looks cool, will give it a try. Thanks!

Falcosc, 2021-05-12:
You should try it without adjustments after a fresh reboot first. Maybe your tunings does trigger some failsafe method from acer.

bagusnl, 2021-05-12:
The thing is that, it only happens on certain workload. Last time I tested it, I need my friend to stream at 1080p60 on discord and I watch, then it would randomly just happen.
I'll try and look if I can make my friend streams again so I can test it.

I just so happen to changed my thermal paste today to much better one (CM MasterGel Maker) so I might have some troubles triggering it again. But we'll see

PR #152 (2021-05-11): Fix Core Temp reading for Dual Core CPUs.
resolves #151
commit 6e7561d1 (2021-05-11): Fix Core Temp reading for Dual Core CPUs.
resolves #151