FlyGoat/RyzenAdj #154: regarding the application of ryzenadj for ryzen 5 5500U

Vijay-Vinith, 2021-05-22:
Hello!
I've recently purchased an acer aspire 7 laptop with the new ryzen 5 5500U Soc with nvidia gtx 1650 gpu
I believe that restricting the cpu tdp tp about 20 watts could provide me with adequate head room for my gpu to boost higher and keep my cpu boost constant and stable but while im trying to apply the temperature and power limits, the app shows that it is unable to apply ryzen adj... Is there any sort of fix for U series cpus of the 5000 series???

Falcosc, 2021-05-22:
Unable to apply is not an error message from ryzenAdj.

Please test your adjustment directly with our CLI and tell us what the real message was.
Execute `ryzenadj -h` first to learn about the usage.

Here is a list of common problems and errors: https://github.com/FlyGoat/RyzenAdj/wiki/FAQ

gompa, 2021-05-22:
pritty sure you are using ryzen controller and haven't set the ryzenadj path

Vijay-Vinith, 2021-05-23:
is there any tutorial on how to set paths and properly apply the correct configs to be able to successfully apply ryzenadj??

Vijay-Vinith, 2021-05-23:
I'm quite new to this stuff and I want an overall performance improvement with better thermals and also i'm quite new to this ryzen controller as well so any tutorials can surely help...

Falcosc, 2021-05-23:
No, not here. We take only functional issues and don't provide support for the ui tools.

closed 2021-05-23 by Falcosc