FlyGoat/RyzenAdj #156: WinRing0 Err: 0x2Unable to get PCI Obj Unable to init ryzenadj, check permission

rajat98, 2021-06-12:
Unable to use any app that uses ryzenadj after the last BIOS update 
Laptop: ASUS ROG ZEPHYRUS G15(2021) model
CPU: Ryzen 7 5800HS

Falcosc, 2021-06-12:
You need to disable security features of your new bios,

Try simple things like disable secure boot first. If disable secure boot doesn't work, you need to find the bios option which does restrict access.

You are starting as administrator, right? Please double-check to make sure that you didn't forget that after your bios update.

rajat98, 2021-06-16:
I have already disabled secure boot, memory integrity and running it as an administrator, yet the issue persists.

Falcosc, 2021-06-17:
Wait a minute, on windows you should get a error string starting with "WinRing0 Err: "

Which version do you use?

Please redownload it and use it via command line.

if you still don't get "WinRing0 Err: " String, I will add some more outputs into this method:
https://github.com/FlyGoat/RyzenAdj/blob/9c548a7159f6c23829031594a1e9003e26214906/lib/osdep_win32.cpp#L31-L60

I suspect it is more related to anti cheat or anti virus software if you have trouble loading the dll.

rajat98, 2021-06-17:
On running installServiceTask.bat as administrator getting "RyzenAdj could not get initialized. Reason can be found inside Powershell. " in a alert box.

Falcosc, 2021-06-17:
Just run ryzenAdj.exe via cmd and copy the full output.

Falcosc, 2021-06-17:
@rajat98 you have an outdated version of ryzenAdj

`WinRing0 Err: 0x2Unable to get` doesn't exist anymore. I missed that.

The new version should output:
`WinRing0 Err: Driver not loaded"

That means that something does block the driver (driver file is available because otherwise it would say `WinrRing0 Err: Driver not found`)

rajat98, 2021-06-17:
Can you tell me where I can find the latest version of ryzenadj.exe or how can I generate it on windows?

rajat98, 2021-06-17:
WinRing0 Err: Driver not loaded
Unable to get PCI Obj, check permission
Unable to init ryzenadj

This is the error I'm receiving.

Falcosc, 2021-06-17:
At least the error handling isn't broken. :)

I am not sure what this driver does during load, normally driver should do as little as possible during initialization.
If the driver complied to best practice, then it is less likely related to you bios update and more likely related to software blocking the driver. But all guesses, I don't know how to debug driver load problems. Because we are not driver developers. If you want to know what is blocking the driver, you need to search for someone who knows about windows driver development.

We can't support you anymore. From here you have to trial and error until you find the problem.

To check if your bios update is related to the issue, I would use a linux live system booted from usb and call ryzenAdj on the linux system.

If linux works, then it is a software issue on your windows system

rajat98, 2021-06-17:
Thanks for the info.

Falcosc, 2021-06-17:
I highly suggest trying linux first, this is much more easy then finding a windows driver developer. Let me know if it works on linux, because if not, then we add bios to the possible cases of this issue.

rajat98, 2021-06-17:
ya, sure.

Falcosc, 2021-06-17:
Just tell me if you need help on the Linux system.

Falcosc, 2021-07-14:
@rajat98 what is the result? We want to change the FAQ if you did confirm this is an issue with your bios or with the os

Chosenplayer100, 2024-06-15:
I know this is 3 years late but i hope you guys can still see this
I am also a person experiencing this problem including in every app
UXTU, Helles Ryzen Test, Ryzen Adj and anything that involves with WinRing0x64, Driver unloaded
Things that i can confirm that are not colliding with this is 
1. Anti Cheat / virus
2. Bios / Windows Update (I even repaired / Reinstalled my windows, to no avail)

but the things i haven't tried yet is if its a 
1. Driver issue
2. Cpu issue
I just pray it isn't because i do not want to uninstall and reinstall again just for it to not work

additional info, in my helles and ryzen adj, the ui is working but the functionality is not, helles ryzen test could read the Cpu Status of its v-core and etc. but however, it is reading NaN, hope this helps to find the error because im worried for my laptop