FlyGoat/RyzenAdj #162: 这个软件打不开，

bigHeadAzuo, 2021-07-21:
ryzenadj 应用程序是灰色的，怎么用？

FlyGoat, 2021-07-21:
请参考GUI软件 https://ryzencontroller.com/，RyzenAdj不是为一般用户设计的，是这些GUI软件的后端。

do-n-khanh, 2021-07-29:
does https://gitlab.com/ryzen-controller-team/ryzen-controller/-/releases/2.3.0 use your v0.8.2 ?

FlyGoat, 2021-08-05:
> does https://gitlab.com/ryzen-controller-team/ryzen-controller/-/releases/2.3.0 use your v0.8.2 ?

yes

closed 2021-08-05 by FlyGoat

do-n-khanh, 2021-08-11:
Thanks. I am using zhen 66 pro 5800u. I tried ryzen-controller setting like below. However, when I do the test, it runs at 28w for a few seconds, and then drops to 25w for 1-2 mins, and then drops to 15 w. Do you know what is wrong with my settings? 

[image]
[image]

Falcosc, 2021-08-11:
> 
> 
> Thanks. I am using zhen 66 pro 5800u. I tried ryzen-controller setting like below. However, when I do the test, it runs at 28w for a few seconds, and then drops to 25w for 1-2 mins, and then drops to 15 w. Do you know what is wrong with my settings?

CPU TDP (W) used the stapm-limit parameter, then this will not change your 15W limit. Read this:
https://github.com/FlyGoat/RyzenAdj/wiki/Supported-Models#sttv2-dependencies-renoir--lucienne--cezanne

And read this: https://github.com/FlyGoat/RyzenAdj/wiki/Options#stapm

And to really understand what is happening you need to read this whole page, it applies to your generation as well.
https://github.com/FlyGoat/RyzenAdj/wiki/Renoir-Tuning-Guide

And don't forget our FAQ:
https://github.com/FlyGoat/RyzenAdj/wiki/FAQ