FlyGoat/RyzenAdj #163: 0.8 divider/multiplier for TDP values for some reason

ioerrorle, 2021-07-21:
When I use this utility on Asus G14 laptop (PM table version 37005, CPU Ryzen 4800HS), setting value for stapm-limit, fast-limit and slow-limit, and then getting it with -i option always results that set value is multiplied to 0.8 of what was provided.
Don't know if it's a laptop-specific problem or CPU-specific.

ioerrorle, 2021-07-21:
Just checked it again and now it works as it supposed to. What the..?
I used G14ControlV2, maybe old ryzenadj executable somehow affected it

ioerrorle, 2021-07-21:
OH! Quick update, it caused by "AMD power slider" option in Windows power plan. It reduces TDP by 20% automatically. Good to know. Issue is closed.

closed 2021-07-21 by ioerrorle

Falcosc, 2021-07-21:
> 
> 
> OH! Quick update, it caused by "AMD power slider" option in Windows power plan. It reduces TDP by 20% automatically. Good to know. Issue is closed.

How did you find this? I had to work around it. There is even a 40% option on the lowest power slider state on battery. Not sure if the 40% one is documented in your source as well.