FlyGoat/RyzenAdj #170: RyzenAdj spits out error on event logger when applying --max-performance or --power-saving

bagusnl, 2021-09-14:
Everything works normally just noticed this during debugging something else unrelated. It also seems like quite slow when applying those 2 variables.

```
Error	9/14/2021 11:43:37 PM	Application Error	1000	(100)
Log Name:      Application
Source:        Application Error
Date:          9/14/2021 11:43:37 PM
Event ID:      1000
Task Category: (100)
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      DESKTOP-*
Description:
Faulting application name: ryzenadj.exe, version: 0.0.0.0, time stamp: 0x60a12713
Faulting module name: ryzenadj.exe, version: 0.0.0.0, time stamp: 0x60a12713
Exception code: 0xc0000005
Fault offset: 0x0000000000001057
Faulting process id: 0x2208
Faulting application start time: 0x01d7a987aa56f83e
Faulting application path: C:\Program Files\RyzenAdj\0.8.2\1\ryzenadj.exe
Faulting module path: C:\Program Files\RyzenAdj\0.8.2\1\ryzenadj.exe
Report Id: d778c312-c61a-4d15-a085-5c3cad4b6295
Faulting package full name: 
Faulting package-relative application ID: 
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" />
    <EventID Qualifiers="0">1000</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>100</Task>
    <Opcode>0</Opcode>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2021-09-14T16:43:37.5839578Z" />
    <EventRecordID>61177</EventRecordID>
    <Correlation />
    <Execution ProcessID="0" ThreadID="0" />
    <Channel>Application</Channel>
    <Computer>DESKTOP-*</Computer>
    <Security />
  </System>
  <EventData>
    <Data>ryzenadj.exe</Data>
    <Data>0.0.0.0</Data>
    <Data>60a12713</Data>
    <Data>ryzenadj.exe</Data>
    <Data>0.0.0.0</Data>
    <Data>60a12713</Data>
    <Data>c0000005</Data>
    <Data>0000000000001057</Data>
    <Data>2208</Data>
    <Data>01d7a987aa56f83e</Data>
    <Data>C:\Program Files\RyzenAdj\0.8.2\1\ryzenadj.exe</Data>
    <Data>C:\Program Files\RyzenAdj\0.8.2\1\ryzenadj.exe</Data>
    <Data>d778c312-c61a-4d15-a085-5c3cad4b6295</Data>
    <Data>
    </Data>
    <Data>
    </Data>
  </EventData>
</Event>
```

Falcosc, 2021-09-14:
SMU is not designed to handle parallel commands. Especially the initialization (if you use ryzenAdj.exe it does always do all the initialization steps) could get upset if something else writes things to the SMU in the same time.

If you talk about Windows Events, my first thought is that calling ryzenAdj.exe during or directly after a power source change or power profile change could be a bit unstable.

2 Workarounds:

1. use a delay to avoid initializing ryzenAdj.exe parallel to SMU commands send by windows power plan.
2. reduce the amount of SMU writes from 40 to 5 by reusing an initialized ryzenadj instance, look at the PowerShell or python example to see how you can reuse an already initialized ryzenAdj instant.

But it is just a wild guess because this is the only thing which comes to my mind. Does this problem description fit your use case?

Please log the output of ryzenAdj.exe during this fault. I can't read anything from this log.

@FlyGoat I don't know how to analyze the responses of a compiled C application well enough, can you do anything with the exception code or fault offset?

bagusnl, 2021-09-15:
> use a delay to avoid initializing ryzenAdj.exe parallel to SMU commands send by windows power plan.

I'm sorry in advance to not giving up more details of what's going in, so here's the situation.
I have a batch script that trigger RyzenAdj once at startup using Task Scheduler with the delay of 1 minute
```
@echo off
set ryzenadjpath="C:\Program Files\RyzenAdj\0.8.2\1\ryzenadj.exe"
%ryzenadjpath% --stapm-limit=20000 --fast-limit=23000 --slow-limit=25000 --tctl-temp=90 --vrm-current=48000 --vrmmax-current=52000 --vrmsoc-current=15000 --vrmsocmax-current=20000 --min-fclk-frequency=500 --max-fclk-frequency=1200 --min-gfxclk=500 --max-gfxclk=1200 --stapm-time=1000 --slow-time=60 --max-performance
```
It does spits an error on this run in Event Viewer, saying that RyzenAdj crashed. It also does when you only do this
```
PS C:\> ryzenadj --max-performance
Sucessfully enable max_performance
PS C:\>
```

After the message "Sucessfully enable max_performance" there is a stop for about 2-3 seconds, presumably due to the crash and Windows Error Report that it triggered (my apologize again for not including this on the first post)
[RyzenAdj_EventLog_APPCRASH.log](https://github.com/FlyGoat/RyzenAdj/files/7165912/RyzenAdj_EventLog_APPCRASH.log)
[Report.wer.log](https://github.com/FlyGoat/RyzenAdj/files/7165913/Report.wer.log)

FlyGoat, 2021-09-16:
@Falcosc What do you mean by fault offset? Catch exception signals?

Falcosc, 2021-09-16:
@bagusnl try the debug build (need some dev tools to execute it)

https://github.com/FlyGoat/RyzenAdj/suites/3742805958/artifacts/91793068

Then we know where it does crash.

bagusnl, 2021-09-16:
@Falcosc 
Here's the command output
```
PS C:\Users\bagusnl_reg\Downloads\Compressed\ryzenadj-win64-debug> .\ryzenadj-debug --max-performance
WinRing0 Err: Driver not found
Unable to get PCI Obj, check permission
Unable to init ryzenadj
PS C:\Users\bagusnl_reg\Downloads\Compressed\ryzenadj-win64-debug> .\ryzenadj-debug --max-performance
SMU_SERVICE REQ_ID:0x18
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
Sucessfully enable max_performance
PS C:\Users\bagusnl_reg\Downloads\Compressed\ryzenadj-win64-debug>
```
I had to rename the executable because I have one on PATH. The first run which it refuses because no WinRing0, it doesn't produce any error on Event Viewer, despite it doesn't actually run. But the 2nd one as expected, does put error on Event Viewer.

Event Viewer output: https://gist.github.com/bagusnl/d11f922487f1d3fca62d8d0e244d66f3

bagusnl, 2021-10-30:
Fixed with #171

closed 2021-10-30 by bagusnl