FlyGoat/RyzenAdj #174: Customize stapm-limit and other values when AC/battery mode

jinwk00, 2021-10-30:
As written in title, I want the stapm-limit to become 60000 instead of default 45000, is it possible to change? If not, are there any alternative that I can set up a script for?

Falcosc, 2021-10-30:
https://github.com/FlyGoat/RyzenAdj/wiki/Options#stapm

- Gets overwritten by SSTv2.
- Only usable if you did disable SSTv2 (Zen2/Zen3)
- Only usable if you don't have SSTv2 (Zen or Zen+)

jinwk00, 2021-10-30:
I mean, when I put my laptop into charger, then stapm/fast/slow automatically becomes 45000, 55000, 48000, is it possible to override this value to something else? And others too, e.g. VRM?

Falcosc, 2021-10-30:
https://github.com/FlyGoat/RyzenAdj#windows-installation

jinwk00, 2021-10-30:
@Falcosc I'm not talking about installation; I am wondering if default values when my AC gets connected or disconnected can be changed or not, even if it means to use a script. Please stay on topic what I am trying to explain.

Falcosc, 2021-10-31:
No ryzenAdj does only apply values over SMU and can not be used to alter your factory power settings, you can only detect the state change and apply the values to overwrite the factory setting every time.

Not sure how Linux does apply the factory values, you may can hook into that: https://github.com/sibradzic/stapmlifier

jinwk00, 2021-10-31:
Thank you, that's the answer I wanted to hear.

closed 2021-10-31 by jinwk00