FlyGoat/RyzenAdj #177: ACPI conflicting values with RyzenAdj

graphcareful, 2021-11-29:
Hi I have found an interesting issue with Ryzen and ACPI but I'm not sure which subsystem or layer is responsible, so this is more of a question then an issue reported. 

I have a Lenovo Thinkpad P14s gen 2 AMD with a Ryzen 7 5850U Pro, running a fresh install of Fedora 35.

```
$ echo '\_SB.PCI0.LPC0.EC0.HKEY.DYTC 0x17d001' > /proc/acpi/call
$ sudo ryzenadj --info 
<Reports ~25 for stapm-limit>
$ cat /sys/firmware/acpi/platform_profile
low-power

$ echo '\_SB.PCI0.LPC0.EC0.HKEY.DYTC 0x12d001' > /proc/acpi/call
$ sudo ryzenadj --info 
<Reports ~11 for stapm-limit>
$ cat /sys/firmware/acpi/platform_profile
performance
```
So pretty much when I write one value to `/proc/acpi/call` i receive conflicting expected values from ACPI and ryzenadj. When `low-profile` is reported `stapm-limit` is high and when `performance` is reported `stapm-limit` is low. Could this be a bug in the DSDT tables?

graphcareful, 2021-11-29:
Upon further investigation i'm wondering if the `thinkpad_acpi` driver could be responsible? https://github.com/torvalds/linux/blob/e26dd976580a6a427c69e6116508dd3d412742e5/drivers/platform/x86/thinkpad_acpi.c#L10052

FlyGoat, 2021-11-30:
Latest `thinkpad-acpi` will handle DYTC with [platform profile](https://www.kernel.org/doc/html/latest/userspace-api/sysfs-platform_profile.html).
So please don't do acpicall on your own, otherwise you're fighting with the driver.

FlyGoat, 2021-11-30:
Also it is known that some vendor has a preference that set lower stapm but higher ppt in performance mode to have a better throttle behavior. Though in your case 11W stapm appears to be too low....

graphcareful, 2021-11-30:
More discussion here https://forums.lenovo.com/t5/Other-Linux-Discussions/T14-AMD-subpar-CPU-performance/m-p/5089829?page=3#5498153

curbengh, 2022-02-02:
[this article](https://www.phoronix.com/scan.php?page=news_item&px=Lenovo-AMD-PP-Not-Working) mentions thinkpad_acpi doesn't support platform profile properly and there is a pending [mainline patch](https://git.kernel.org/pub/scm/linux/kernel/git/pdx86/platform-drivers-x86.git/commit/?h=for-next&id=836f35f79153ce09d813c83f341dba4481996966) to disable the support while Lenovo is working on reintroduce it.

dylangerdaly, 2022-06-24:
I assume the PCIe device RyzenAdj is interacting with is the same thing that allows users to overclock via userland/Windows?

I've never really understood fully how ACPI falls into this, I assumed ACPI is where vendors set default for all of the power/performance related settings?

graphcareful, 2022-06-24:
I actually looked into this, made some posts on the Lenovo forums. A kernel committer responded with a comment how the latest AMD ThinkPads just don't have the driver support upstream, so the values reported by ACPI are not correct.

closed 2022-06-24 by graphcareful

dylangerdaly, 2022-06-26:
Is that so bad? I think I would prefer not utilize ACPI at all and _just_ change these settings via RyzenAdj.

Does anyone know where the default values are set in ACPI? For Temp limits etc?

FlyGoat, 2022-06-28:
> I assume the PCIe device RyzenAdj is interacting with is the same thing that allows users to overclock via userland/Windows?
> 
> I've never really understood fully how ACPI falls into this, I assumed ACPI is where vendors set default for all of the power/performance related settings?

Nah actually it’s UEFI bios code who is setting the default values on boot. Afterwards vendors may try to overlay those value in their custom ACPI codes (DYTC for Lenovo) or EC queries. And RyzenAdj is talking to SMU in the same way.

dylangerdaly, 2022-06-28:
Do you happen to know the DXE name Lenovo uses? I'd like to have a look at a Ryzen 6000 laptop I've ordered, I've already extracted the ACPI tables from the UEFI Update.

_Offtopic: From what I can tell I can disable the dGPU via an _OFF ACPI method, I'm guessing that just flips an EC GPIO attached to a MOSFET on the 12V rail for the dGPU._

So is UEFI setting these values the same way RyzenAdj is? The PCIe device?

FlyGoat, 2022-06-28:
> Do you happen to know the DXE name Lenovo uses? I'd like to have a look at a Ryzen 6000 laptop I've ordered, I've already extracted the ACPI tables from the UEFI Update.
If I recall correctly, the default value is set in `AmdNbioSmuV10Dxe`, it will get vendor's default settings from UEFI Pcd database and apply to SMU.

> 
> _Offtopic: From what I can tell I can disable the dGPU via an `_OFF` ACPI method, I'm guessing that just flips an EC GPIO attached to a MOSFET on the 12V rail for the dGPU._
> 
> So is UEFI setting these values the same way RyzenAdj is? The PCIe device?
Partially yes. There are some complicated "Table Transfer" process which directly transfer settings from memory to SMU but they are also initiated by SMU service requests.

graphcareful, 2022-06-28:
Another update here. Just updated my laptop to 5.18, seems like Marks patches have worked !

dylangerdaly, 2022-06-29:
> it will get vendor's default settings from UEFI Pcd database and apply to SMU.

Okay this sounds super interesting, are these the UEFI settings that "THEY DON'T WANT YOU TO KNOW", and by that I mean don't show via their UEFI Settings Interface? I remember reading about a Utility that let you open and configure a database, I don't remember if it was PCD.

Knowing this would be awesome for me as I'd like to make sure TSME (Transparent Memory Encryption) is enabled on my non-PRO AMD SKU CPU.

Would you happen to know how I can locate the PCD Database with UEFI Tool?

> I'd like to just click between balanced/performance/battery and also let the OS auto-modify these parameters depending if the device is plugged in or not, etc.

When running under a Xen Hypervisor (QubesOS), on AC it'll automatically go into "Performance" mode and spike up to ~85C doing nothing, it's normal for ACPI to be broken under a Xen Hypervisor, but you're right, the normal user would prefer ACPI take care of everything.

FlyGoat, 2022-07-01:
@dylangerdaly I don’t have much idea about how does Pcd database work on Insyde bios either.
Tried to reverse engineering based on EDK2’s code but it seems like the implementation is totally different.

dylangerdaly, 2022-09-09:
Hi all,

I'm currently trying to raise the temp limit on my Ryzen 4000 series laptop, however upon using RyzenAdj, I can see my changes are immediately being overridden.

I assume this is an ACPI issue, can anyone please advise? Thank you.