FlyGoat/RyzenAdj #178: I can't change my maximum TDP anymore

VuH0ngQuang, 2021-12-16:
After I upgrade my BIOS from MSI, I can't change my maximum TDP anymore. It's stuck at 45W and can't make it run higher. But when I limit TDP it's still working
My laptop is Bravo 15 2021
Ryzen 7 5800H
BIOS ver: E158KAMS.10E
EC ver: 158KEMS1.106
And all drivers all newest

FlyGoat, 2021-12-16:
My personal suggestion is just downgrade your BIOS. We don't have anyway to debug AMD SMU internals.

VuH0ngQuang, 2021-12-17:
MSI doesn't offer an older BIOS version on their website. So I think I'm stuck with this problem :(

VuH0ngQuang, 2021-12-27:
> My personal suggestion is just downgrade your BIOS. We don't have anyway to debug AMD SMU internals.

I'm just downgrade my BIOS but still not working. Is there any way to detected the error?

angelokimhui, 2021-12-27:
> After I upgrade my BIOS from MSI, I can't change my maximum TDP anymore. It's stuck at 45W and can't make it run higher. But when I limit TDP it's still working My laptop is Bravo 15 2021 Ryzen 7 5800H BIOS ver: E158KAMS.10E EC ver: 158KEMS1.106 And all drivers are newest

If you enabled power_saving in ryzen adj, it usually limits the STAPM TDP and make sure Windows is not in powersaving. Set the slider to max performance.

VuH0ngQuang, 2021-12-28:
> If you enabled power_saving in ryzen adj, it usually limits the STAPM TDP and make sure Windows is not in powersaving. Set the slider to max performance.

Actually, I'm using Ryzen controller, and i don't enable power saving at all, all is in the performance mode