FlyGoat/RyzenAdj #179: 2.5.4 unstable in Windows 11

idkillapuppyforyou, 2021-12-29:
Black screens soon after startup. Unable to acquire dump. Also what I consider minor in 2.5.3 doesn't queue up in start-up so manual start necessary even when modifying start-up behavior under properties\compatibilities. I'm happy with 2.5.3; works great, very stable, does everything it should, like the addition of stats. You have my gratitude. Thanks Guys!

Device name	GIMP
Processor	AMD Ryzen 7 3700U with Radeon Vega Mobile Gfx     2.30 GHz
Installed RAM	20.0 GB (17.9 GB usable)
Device ID	E42AF95A-7ABA-4300-BE5B-CF27395394BC
Product ID	00330-80000-00000-AA252
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display
Edition	Windows 11 Pro
Version	21H2
Installed on	‎10/‎7/‎2021
OS build	22000.376
Experience	Windows Feature Experience Pack 1000.22000.376.0

Falcosc, 2021-12-29:
I guess you are at the wrong place, we don't have any release which does match your description:

https://github.com/FlyGoat/RyzenAdj/releases

JamesCJ60, 2022-02-15:
Hi,
I believe you are talking in regards to Ryzen Controller with this issue, if so here's the link to the correct place to report this issue: https://gitlab.com/ryzen-controller-team/ryzen-controller/-/issues

closed 2022-02-20 by JamesCJ60