FlyGoat/RyzenAdj #181: linux 5850U pro issues

installgentoo, 2022-01-13:
Okay, this cpu is notably crap on linux. There's a phoronix article about lenovo machines with it being stuck on 15w, and if i run my L15 it drops to 12w and sits there forever. Default max skin temp is 48, which get tripped in seconds.

If you try to adjust judging by displayed values something still gets tripped and throws you to 12w. Note that all the while cputemp doesn't go beyond 60C.

I was able to stabilize at 25W for tens of seconds by setting all 4 temp limits to 80C(though something still trips the cpu and resets all limits, which then trips the limits and sends it back to 12w). So perhaps temps are in the wrong order in ryzenadj?

Besides reporting this apparent bug with this specific cpu i'd very much appreciate tips on figuring out what trips the cpu to reset the limits, besides temp.

installgentoo, 2022-01-13:
Also stapm-limit on 5850 maxes out at 25w which seems like an arbitrary restriction. On 25w my L15 sustains 69C with fan on half the speed. other values max out too at 37 and 48 w.

On my 4500u i can push wattage to absolutely arbitrary values and from cold start it does sustain 30w for a couple seconds before being throttled by temperature. I didn't experiment with temps above 80C and with high wattages in general as to not set the computer on fire.

So is the 25w restriction something in zen3 cpu which can't be overridden, or just bios limit?

mvdan, 2022-01-13:
@installgentoo I'm on a p14s gen2 5850u, and I also struggle to make it consume more than 25W, even for short bursts. I'm using this config:

    ryzenadj --stapm-limit=30000 --fast-limit=30000 --slow-limit=25000 --tctl-temp=90 --apu-skin-temp=55 --max-performance

I don't seem to be reaching any of the temp limits, even under sustained load; tctl-temp only reaches 76, and apu-skin-temp only reaches 40. All cores can reach 3.2GHz briefly, but they settle at 2.9GHz after a little while, all the while consuming 25W. I should note that ryzenadj still helps; it used to take a couple of seconds to properly ramp up, and it used to drop below 25W under sustained load.

At least as per the spec, I should be able to get an all-core boost to 3.4GHz, but I haven't been able to. I'm also fairly sure this p14s should be able to sustain load above 25W; a CPU temp of just under 80C isn't particularly high. Here's my `ryzenadj -i` after two minutes of all-core load and my config above, if you want to compare notes:

```
# ryzenadj -i
CPU Family: Cezanne
SMU BIOS Interface Version: 18
Version: v0.8.3 
PM Table Version: 400005
|        Name         |   Value   |      Paramter      |
|---------------------|-----------|--------------------|
| STAPM LIMIT         |    25.001 | stapm-limit        |
| STAPM VALUE         |    25.013 |                    |
| PPT LIMIT FAST      |    30.000 | fast-limit         |
| PPT VALUE FAST      |    25.640 |                    |
| PPT LIMIT SLOW      |    25.000 | slow-limit         |
| PPT VALUE SLOW      |    25.003 |                    |
| StapmTimeConst      |     1.000 | stapm-time         |
| SlowPPTTimeConst    |     5.000 | slow-time          |
| PPT LIMIT APU       |    37.500 | apu-slow-limit     |
| PPT VALUE APU       |    25.003 |                    |
| TDC LIMIT VDD       |    44.000 | vrm-current        |
| TDC VALUE VDD       |    23.858 |                    |
| TDC LIMIT SOC       |    13.000 | vrmsoc-current     |
| TDC VALUE SOC       |     1.643 |                    |
| EDC LIMIT VDD       |    95.000 | vrmmax-current     |
| EDC VALUE VDD       |    49.123 |                    |
| EDC LIMIT SOC       |    17.000 | vrmsocmax-current  |
| EDC VALUE SOC       |     0.000 |                    |
| THM LIMIT CORE      |    90.000 | tctl-temp          |
| THM VALUE CORE      |    77.141 |                    |
| STT LIMIT APU       |    55.000 | apu-skin-temp      |
| STT VALUE APU       |    44.712 |                    |
| STT LIMIT dGPU      |     0.000 | dgpu-skin-temp     |
| STT VALUE dGPU      |     0.000 |                    |
| CCLK Boost SETPOINT |    50.000 | power-saving /     |
| CCLK BUSY VALUE     |   100.000 | max-performance    |
```

One last bit of info: this is on Linux 5.16 with the latest UEFI firmware available from Lenovo. It seems like Linux 5.17 will bring AMD's new P-State driver; it's unclear to me whether that will help or not: https://www.phoronix.com/scan.php?page=news_item&px=Linux-5.17-Early-Features

mvdan, 2022-01-13:
I should also note that I also see max values for some limits; for instance:

```
$ ryzenadj --stapm-limit=48000 --fast-limit=48000 --slow-limit=48000
# ryzenadj -i | grep -- -limit
| STAPM LIMIT         |    25.001 | stapm-limit        |
| PPT LIMIT FAST      |    48.000 | fast-limit         |
| PPT LIMIT SLOW      |    37.501 | slow-limit         |
| PPT LIMIT APU       |    37.501 | apu-slow-limit     |
```

Finally, note that [this recent review](https://www.notebookcheck.net/Lenovo-ThinkBook-13s-G3-AMD-laptop-in-review-Subnotebook-with-fast-Ryzen-CPU.591754.0.html) featuring a 5800u on a ThinkBook reports a burst to 30W and then settling at 25W, but it also mentions:

> You can use the optional Maximum Performance mode for eking out even more performance. Here, the TDP increases up to 48 W, but then it gets limited to 19 W after a few minutes. At least for short periods, the ThinkBook 13s G3 can thus offer the CPU performance of full-fledged mobile workstations with 45 W CPUs from Intel. 

I'd love to get the same, but on Linux :) Presumably a P14s should be able to reach 45W, given that it's larger than that 13" thin laptop.

mvdan, 2022-01-13:
https://github.com/FlyGoat/RyzenAdj/wiki/Supported-Models#custom-value-range-for-zen3 is also worth a mention, as it talks about these problems. I haven't tried its workarounds yet.

installgentoo, 2022-03-09:
still a thing on 9.0.0, though now cpu doesn't drop to 12 as fast. so i guess something in the linux kernel impacts how fast this reset functionality kicks in?

Falcosc, 2022-04-18:
@mvdan did already found the correct wiki entry
.001 W values are the indicator for the new vendor lock of limits.

installgentoo, 2022-04-18:
Can vendor lock be bypassed within ryzenadj?

I tried adjusting locked values via https://wiki.archlinux.org/title/DSDT , didn't seem to work.

Falcosc, 2022-04-19:
No, that's the point of the new vendor lock. To limit the capabilities of the undocumented power adjustments API.

installgentoo, 2022-05-01:
I tried this again, just outright deleted all the ALIB calls in DSDT. As a result my cpu stopped resetting, but it's still locked to 25W STAMP. I tried overriding 0x05 with 30W, but it's still locked to 25W.

Anyone has any ideas as to what may be limiting STAMP besides bios?

gstjee, 2026-08-25:
@installgentoo did you find any solution ?