FlyGoat/RyzenAdj #183: Fans not ramping up anymore

sigygtdvcx, 2022-01-24:
My laptop's fans no longer respond to increase in CPU/GPU temperature. I tried changing the TDP and Temp limit and even after unchecking and restarting, my fans will not ramp up. The CPU sticks to the settings but instead of spinning the fans above idle, it throttles and lowers the frequency. System restore doesn't help. I've had this before and once I reinstalled windows but the second time, I remember to have changed some random settings to high values, reset, then some low values then reset again (or something similar) and it somehow brought back the fan control. Now I can't figure anything out and there is no default button.
Lenovo Ideapad Gaming Ryzen 5 4600H

sigygtdvcx, 2022-01-26:
fixed: flash and older bios ugh

FlyGoat, 2022-01-29:
We have nothing to do with fans....

closed 2022-01-29 by FlyGoat