FlyGoat/RyzenAdj #188: Report AGESA version

drws, 2022-02-19:
There is a shortage of Linux tools that could report AMD's AGESA (_AMD Generic Encapsulated Software Architecture_) version. Windows-based [HWiNFO](https://www.hwinfo.com/) and [AIDA64](https://www.aida64.com/) reportedly [have the ability](https://www.reddit.com/r/Amd/comments/7mdn6n/how_to_identify_agesa_code_running_on_your_system/), but I couldn't find any Linux-based solution.

FlyGoat, 2022-02-20:
I peronally have no idea where this information could be gathered.
Anyone did reverse engineering?

patrickschur, 2022-02-20:
I don't know how HWiNFO or AIDA64 determines the AGESA version. However, what I do know is that these tools can't always determine the current version. At least they don't work for me. **I guess** they are using some kind of table which contains the AGESA and the SMU firmware version. But maintaining this kind of table is not fun and would require regular updates.

mirh, 2022-04-14:
https://bugzilla.kernel.org/show_bug.cgi?id=201473
You have to quite brutally scan the first megabytes of memory.

BVollmerhaus, 2025-02-16:
I hope this won't come across as shameless self-promotion, but I just published a standalone CLI tool (and Rust library) for exactly this purpose: [agesafetch](https://crates.io/crates/agesafetch)

As @mirh said, the AGESA version can be found somewhere in extended memory (> 1 MiB) via a brute-force search. From my testing, it is always contained in at least one `Reserved` memory region, usually the first one at that. The address ranges and types of known regions can be extracted from `/proc/iomem`, and raw memory can then be read from `/dev/mem` (assuming elevated privileges; some sections are still inaccessible despite that).

The signature indicating the start of the AGESA version in memory differs between AGESA v9 (>=`17h`, i.e. Zen) and AGESA v5 (<`17h`). The former one is `AGESA!V9` (though I'm not sure if it is always `9`, for instance on Epyc), and the latter – which my tool does not yet support – is `!!AGESA ` per the [AMD reference documentation](https://www.amd.com/content/dam/amd/en/documents/processor-tech-docs/specifications/44065_Arch2008.pdf).

In short, you have to sequentially search all `Reserved` memory regions above `0x00100000` for the byte sequence `AGESA!V9`, which is followed by a null byte, **the actual version**, and another closing null byte.

It's certainly not as pretty as having the kernel provide a native interface or AGESA being included in the SMBIOS data (who knows why it isn't), but it **is** possible to obtain it from user space with some effort.

Hope this helps a little. 🙂 Thanks to @mirh for your prior research.

@FlyGoat Feel free to use my implementation for reference.