FlyGoat/RyzenAdj #192: --power-saving while on AC

acoul, 2022-04-23:
Greetings,

I have a Gateway BRIX 4300U and I would like to lower as much as possible the consumption of this system

here is the info output:

`CPU Family: Renoir
SMU BIOS Interface Version: 17
Version: v0.8.3 
PM Table Version: 370003
|        Name         |   Value   |      Paramter      |
|---------------------|-----------|--------------------|
| STAPM LIMIT         |    12.000 | stapm-limit        |
| STAPM VALUE         |     4.589 |                    |
| PPT LIMIT FAST      |    12.000 | fast-limit         |
| PPT VALUE FAST      |     4.630 |                    |
| PPT LIMIT SLOW      |    12.000 | slow-limit         |
| PPT VALUE SLOW      |     4.491 |                    |
| StapmTimeConst      |   275.000 | stapm-time         |
| SlowPPTTimeConst    |    10.000 | slow-time          |
| PPT LIMIT APU       |    12.000 | apu-slow-limit     |
| PPT VALUE APU       |     0.000 |                    |
| TDC LIMIT VDD       |    33.000 | vrm-current        |
| TDC VALUE VDD       |     1.183 |                    |
| TDC LIMIT SOC       |    13.000 | vrmsoc-current     |
| TDC VALUE SOC       |     2.254 |                    |
| EDC LIMIT VDD       |    50.000 | vrmmax-current     |
| EDC VALUE VDD       |    12.054 |                    |
| EDC LIMIT SOC       |    17.000 | vrmsocmax-current  |
| EDC VALUE SOC       |     0.000 |                    |
| THM LIMIT CORE      |   100.000 | tctl-temp          |
| THM VALUE CORE      |    46.717 |                    |
| STT LIMIT APU       |     0.000 | apu-skin-temp      |
| STT VALUE APU       |     0.000 |                    |
| STT LIMIT dGPU      |     0.000 | dgpu-skin-temp     |
| STT VALUE dGPU      |     0.000 |                    |
| CCLK Boost SETPOINT |    95.000 | power-saving /     |
| CCLK BUSY VALUE     |    42.231 | max-performance    |`

best regards

Falcosc, 2022-04-24:
I use the --power-saving option on Renoir. Works great, but only lowers idle power consumption and does only make a difference if you have some bad background applications.

acoul, 2022-04-26:
thank you for the feedback

FWIW, I found the following quite informative for my quest on minimizing power consumption on non-battery systems: https://github.com/kevinlekiller/amdctl/issues/20