FlyGoat/RyzenAdj #193: Rembrandt Binaries

AndreiLux, 2022-04-28:
Would you be able to update the Win binaries with the ID 68 fix on Rembrandt? I'm looking to setup MSVC but it takes a while...

FlyGoat, 2022-04-28:
You can download from artifacts of action CI run.
https://github.com/FlyGoat/RyzenAdj/actions/runs/2174201297

AndreiLux, 2022-04-29:
I ended up compiling myself before I saw your reply.

Unfortunately it looks like currently the Rembrandt patches aren't sufficient; the fast/slow PL don't work - while they're being written they have no effect on the behaviour of the device.

I'm running on a 6800HS on the Flow X13. The device does have the Armory Crate PL controls, please let me know if I can help seeing what has changed in the structure so that we can get working SMU controls.

AndreiLux, 2022-04-29:
The -i option doesn't work - I would have posted it otherwise.

`CPU Family: Rembrandt
SMU BIOS Interface Version: 16
Version: v0.8.3
request_table_ver_and_size is not supported on this family
Unable to init power metric table: -1, this does not affect adjustments because it is only needed for monitoring.`

Can you quick guide me what you want through uProf?

FlyGoat, 2022-04-29:
Ahhh, that means the interface is totally unknown for us :-(

AndreiLux, 2022-04-29:
I figured out what's happening; you can't use any adjustments when the laptop is on any of the predefined power profiles (Windows/Silent/Performance/Turbo) as the limits you set get ignored. Only when in Armoury Crate manual mode does it use those pre-defined settings, and only in that mode does it follow external settings such as via RyzenAdj.

This is in contrast to the Cezanne Flow X13 where it does follow any adjustment regardless of power profile. I haven't tried uninstalling Armoury Crate to see if the limitation can be removed.

closed 2022-04-29 by AndreiLux