FlyGoat/RyzenAdj #196: Support for Vangogh - Steam Deck

brycesub, 2022-05-11:
`(1)(deck@steamdeck ~)$ sudo ./ryzenadj  -i
CPU Family: Vangogh
SMU BIOS Interface Version: 2
Version: v0.10.0
request_table_ver_and_size is not supported on this family
Unable to init power metric table: -1, this does not affect adjustments because it is only needed for monitoring.
(255)(deck@steamdeck ~)$ sudo ./ryzenadj  --dump-table
request_table_ver_and_size is not supported on this family
Unable to init power metric table: -1, this does not affect adjustments because it is only needed for monitoring.`

How can I help facilitate support for the Vangough line of mobile processors?  Tweaking power management is of interest to the Steam Deck community.

Falcosc, 2022-07-29:
Coreboot, Ryzen SMU and Zenpower projects don't support power metric table for Rembrand and Vangogh either. So it is unknown how to get the data because nobody knows which command does expose the dram adress and which command does refresh the data.

But interestingly https://github.com/sbski/Renoir-Mobile-Tuning does not exclude these generations and just execute the commands from the previous generation https://github.com/sbski/Renoir-Mobile-Tuning/blob/29c2fd04fa32b8643bad949f9c09c75cf9626c41/renoir_tuning_utility/renoir_tuning_utility/RMT.cs#L145-L152
Maybe have a look if this is able to collect the table values.

agluszyk, 2022-09-13:
@Falcosc do you think that such method is worth trying https://github.com/agluszyk/RyzenAdj/tree/steamdeck-gfxclk ?

Falcosc, 2022-09-15:
I cannot verify this code without the hardware, but if your linked code works, then you can create a pull request

wzzt, 2022-09-23:
I tried building the above with MSVC/Clang and running it on Windows 11 on the Steam Deck. The SMU always rejects set_max_gfxclk_freq outright, and while ryzenadj consistently reports success when setting the minimum frequency, it appears to have no effect on clock speed or gaming performance regardless of what value it's set to.