FlyGoat/RyzenAdj #197: Support for Rembrandt

jbdrthapa, 2022-05-20:
sudo ryzenadj -i
CPU Family: Rembrandt
SMU BIOS Interface Version: 16
Version: v0.8.3 
request_table_ver_and_size is not supported on this family
Unable to init power metric table: -1, this does not affect adjustments because it is only needed for monitoring.

Jalim19, 2022-07-25:
I also have this problem. But it seems doesn't affect controlling the limitation of power.

Falcosc, 2022-07-29:
Coreboot, Ryzen SMU and Zenpower projects don't support power metric table for Rembrand and Vangogh either. So it is unknown how to get the data because nobody knows which command does expose the dram adress and which command does refresh the data.

But interestingly https://github.com/sbski/Renoir-Mobile-Tuning does not exclude these generations and just execute the commands from the previous generation https://github.com/sbski/Renoir-Mobile-Tuning/blob/29c2fd04fa32b8643bad949f9c09c75cf9626c41/renoir_tuning_utility/renoir_tuning_utility/RMT.cs#L145-L152
Maybe have a look if this is able to collect the table values.