FlyGoat/RyzenAdj #200: [bug?] _do_adjust_psmu(OPT) never gets called

Etaash-mathamsetty, 2022-06-11:
```
EXP int CALL set_oc_volt(ryzen_access ry, uint32_t value) {
	switch (ry->family)
	{
	case FAM_LUCIENNE:
	case FAM_RENOIR:
	case FAM_CEZANNE:
	case FAM_REMBRANDT:
		_do_adjust(0x33);
		_do_adjust_psmu(0x1b);
		break;
	}
	return ADJ_ERR_FAM_UNSUPPORTED;
}`
```
This might be a mistake in me reading the code, but I doubt it.
Basically, _do_adjust is a macro that returns a value once its done executing, so how is do_adjust_psmu ever going to get called in this scenario?

FlyGoat, 2022-06-11:
Yep can confirm it's a problem.

@JamesCJ60 what was your intention?

JamesCJ60, 2022-06-11:
An issue was encountered when testing on desktop APUs when using the CLI and dll versions of RyzenAdj. 

One would only accept PSMU and other one only accept the normal MP1 _do_adjust for the overclocking and voltage control commands. 

This was done so they both worked regardless of which RyzenAdj implementation is used. Without it, only one was be able to apply the changes for these options.

Etaash-mathamsetty, 2022-06-13:
> An issue was encountered when testing on desktop APUs when using the CLI and dll versions of RyzenAdj.
>
> One would only accept PSMU and other one only accept the normal MP1 _do_adjust for the [...]

how does that work then? becuase if only _do_adjust is going to be called, _do_adjust_psmu will never get called no matter what (based on what I read)

Etaash-mathamsetty, 2022-06-13:
I think fixing this could also fix 
#198 
but that might be something else entirely

FlyGoat, 2022-08-07:
Finally caught some time to look into this.
@Etaash-mathamsetty @JamesCJ60 could you please review and test commit on [err_handling](https://github.com/FlyGoat/RyzenAdj/tree/err_handling) branch?

If you would like to have a Windows executable here is CI output: https://github.com/FlyGoat/RyzenAdj/actions/runs/2812559586

Thanks.

Etaash-mathamsetty, 2022-08-07:
> Finally caught some time to look into this. @Etaash-mathamsetty @JamesCJ60 could you please review and test commit on [err_handling](https://github.com/FlyGoat/RyzenAdj/tree/err_handling) branch? [...]
none of the OC functions work and GFX CLK broke after a bios update, but looking over the source code, everything seems fine

JamesCJ60, 2022-08-07:
> > Finally caught some time to look into this. @Etaash-mathamsetty @JamesCJ60 could you please review and test commit on [err_handling](https://github.com/FlyGoat/RyzenAdj/tree/err_handling) branch?
> > If you would like to have a Windows executable here is CI output: https://github.com/FlyGoat/RyzenAdj/actions/runs/2812559586
> > Thanks.
> 
> none of the OC functions work and GFX CLK broke after a bios update, but looking over the source code, everything seems fine when running it, the PSMU and SMU stuff are both being tried

OC functions only work on 4000+G APUs

Etaash-mathamsetty, 2022-08-07:
> OC functions only work on 4000+G APUs

Oh really, i thought they work on all 4000 series apus including mobile ones, might want to add a note about it

closed 2022-12-12 by FlyGoat