FlyGoat/RyzenAdj #207: Not supported mesagges showing on everything I do, plus can't use -i or --dump-table on nevest version

cronyakatsuki, 2022-08-06:
Today I upgraded my ryzenadj version on my arch system and saw that this shows up after everything I ran except `ryzenadj -h`
```
set_coall is not supported on this family
set_coper is not supported on this family
set_cogfx is not supported on this family
```

## Examples

> ryzenadj -i
```
CPU Family: Picasso
SMU BIOS Interface Version: 20
Version: v0.11.0 
set_coall is not supported on this family
set_coper is not supported on this family
set_cogfx is not supported on this family
```
> ryzenadj --dump-table
```
set_coall is not supported on this family
set_coper is not supported on this family
set_cogfx is not supported on this family
```
> ryzenadj **settings options**
```
Sucessfully set stapm_limit to 10000
Sucessfully set fast_limit to 12000
Sucessfully set slow_limit to 11000
Sucessfully set slow_time to 30
Sucessfully set stapm_time to 300
Sucessfully set tctl_temp to 60
Sucessfully set vrmmax_current to 30000
Sucessfully enable max_performance
set_coall is not supported on this family
set_coper is not supported on this family
set_cogfx is not supported on this family
```

## What I expect
I don't mind the message but I think it would be nice to atleast get the output from `-i` and `--dump-table` options.

lenuswalker, 2022-08-06:
I too have this issue.

FlyGoat, 2022-08-07:
Thanks for the report, will investigate.

FlyGoat, 2022-08-07:
The first issue should be resolved, could you please check last commit on master?

FlyGoat, 2022-08-07:
Will release v0.11.1 after receive test report.

cronyakatsuki, 2022-08-07:
Work's normally now. Tnx for a fast fix.

closed 2022-08-07 by cronyakatsuki

lenuswalker, 2022-08-07:
Thanks a lot for the fast fix 👍🏿