FlyGoat/RyzenAdj #210: 6800U is doesn't work

danyi, 2022-08-16:
I used the latest version, V0.11.1.
But it doesn't work
This issue causes me to be unable to adjust on SteamOS

ryzenadj.exe -i
CPU Family: Rembrandt
SMU BIOS Interface Version: 18
Version: v0.11.1
request_table_ver_and_size is not supported on this family
Unable to init power metric table: -1, this does not affect adjustments because it is only needed for monitoring.

ryzenadj.exe --dump-table
request_table_ver_and_size is not supported on this family
Unable to init power metric table: -1, this does not affect adjustments because it is only needed for monitoring.

pastaq, 2022-08-16:
A little more background on the issue: In HandyPT I use dump-table to read the current values to set the initial level of the UI and to validate the settings took accurately. This ensures that I'm not constantly spamming the system with TDP settings and ensures I can let the user know if another tool/setting/mode has changed the TDP as well. The --stapm-limit, --fast-limit, and --TCTL-temp flags all appear to set, but cannot be validated with -i or -dump-table in Linux or Windows. Is this an oversight, or is there a technical limitation that prevents the Rembrandt series from exposing the entire settings table?

pastaq, 2022-08-16:
Sweet. Do you have an example/general outline of what to do? I've never used uProf before.

This seems to be the correct package:
https://aur.archlinux.org/packages/amduprof

pastaq, 2022-08-16:
Installed uProf and started geting familiar with it. What info do you need? My laptop is a 4800U but I'd like to write a step by step for the testers with the 6000 series to make it a bit more simple to get the data we need.

pastaq, 2022-08-23:
> Hmm, it just means that nobody had reverse engineering this table version (which can be affected by processor model and AGESA rev).
> 
> Let's see if anybody will put effort on it. We should be able to reverse it from uProf.

I have two users standing by with 6000 series chips. What uProf data is required for this?

ruineka, 2022-08-26:
> Hmm, it just means that nobody had reverse engineering this table version (which can be affected by processor model and AGESA rev).
> 
> Let's see if anybody will put effort on it. We should be able to reverse it from uProf.

I'm pretty sure you'd need the information from uProf controllers am I correct?

dylangerdaly, 2022-09-07:
I have a 6900HS, I'm happy to take a look at reverse engineering whatever is required to get this working, where should I start looking?

patrickschur, 2022-09-07:
We are already working on support for Rembrandt. The problem is that none of us has a device with such an APU and that makes it harder for us to add and test the support for Rembrandt. If someone wants to help, could he or she please run the following script?

https://gist.github.com/patrickschur/4645f81601fe6f8ff16c7a90e848d7cb

Just download the script and run the following command:

```sh
sudo python3 pmtable.py
```

Please run the script two times and let us know the result of both executions.

Okazakee, 2022-09-08:
[image]

This is what happens with that script

dylangerdaly, 2022-09-08:
Hm, I don't have a kernel that supports `/dev/mem`

```
table base address: 0x3e300000
table version: 0x450004
transfer table: [0, 8, 0, 0, 0]
Traceback (most recent call last):
  File "pmtable.py", line 61, in <module>
    print(f"{i:04X}: {struct.unpack(f, os.read(devmem, 4))[0]}")
PermissionError: [Errno 1] Operation not permitted
```

Is there any way around this? If not I can compile a kernel with /dev/mem support.

patrickschur, 2022-09-08:
As far as I know there is no way around this (at least not from userland).

ruineka, 2022-09-08:
> As far as I know there is no way around this (at least not from userland).

I attempted this on Pop OS and I got the same issue. Does this work on Arch Linux?

patrickschur, 2022-09-08:
Yes, if you compile the kernel with `/dev/mem` support (see [CONFIG_DEVMEM](https://cateee.net/lkddb/web-lkddb/DEVMEM.html) and [CONFIG_STRICT_DEVMEM](https://cateee.net/lkddb/web-lkddb/STRICT_DEVMEM.html)).

ruineka, 2022-09-08:
> Yes, if you compile the kernel with `/dev/mem` support (see [CONFIG_DEVMEM](https://cateee.net/lkddb/web-lkddb/DEVMEM.html) and [CONFIG_STRICT_DEVMEM](https://cateee.net/lkddb/web-lkddb/STRICT_DEVMEM.html)).

Cool, I'll compile the kernel now and try to get you the information.

GreyXor, 2022-09-09:
I'll have a 6850U soon and I'll be happy to help in this

hh1599, 2022-09-13:
> We are already working on support for Rembrandt. The problem is that none of us has a device with such an APU and that makes it harder for us to add and test the support for Rembrandt. If someone wants to help, could he or she please run the following script?
> 
> https://gist.github.com/patrickschur/4645f81601fe6f8ff16c7a90e848d7cb
> 
> Just download the script and run the following command:
> 
> ```shell
> sudo python3 pmtable.py
> ```
> 
> Please run the script two times and let us know the result of both executions.

I have a 6800u device coming next month, but i can borrow a 6800H now.  If I run the script on a 6800H will it help you for the U version?

hh1599, 2022-09-14:
> Yes, if you compile the kernel with `/dev/mem` support (see [CONFIG_DEVMEM](https://cateee.net/lkddb/web-lkddb/DEVMEM.html) and [CONFIG_STRICT_DEVMEM](https://cateee.net/lkddb/web-lkddb/STRICT_DEVMEM.html)).

so just to confirm, 

`config DEVMEM
	bool "/dev/mem virtual device support"
	default y`

in "drivers/char/Kconfig" should be left as is.  

And,

`config STRICT_DEVMEM
	bool "Filter access to /dev/mem"
	depends on MMU && DEVMEM
	depends on ARCH_HAS_DEVMEM_IS_ALLOWED || GENERIC_LIB_DEVMEM_IS_ALLOWED
	default y if PPC || X86 || ARM64`

should be changed to `default n if PPC || X86 || ARM64` in "lib/Kconfig.debug"

patrickschur, 2022-09-14:
@hh1599 `STRICT_DEVMEM` is optional and only needed if you want to restrict the access. Here is my config:

```
CONFIG_DEVMEM=y
CONFIG_ARCH_HAS_DEVMEM_IS_ALLOWED=y
CONFIG_STRICT_DEVMEM=y
# CONFIG_IO_STRICT_DEVMEM is not set
```

You can get the config using `grep DEVMEM /lib/modules/$(uname -r)/build/.config`.

ruineka, 2022-09-15:
> @hh1599 `STRICT_DEVMEM` is optional and only needed if you want to restrict the access. Here is my config:
>
> ```
> CONFIG_DEVMEM=y
> CONFIG_ARCH_HAS_DEVMEM_IS_ALLOWED=y
> CONFIG_STRICT_DEVMEM=y
> # [...]

I tested a build using this config here. https://github.com/ruineka/linux/releases/download/v6.0-rc5-asus-fix/config 
I get the same `  File "/home/gamer/Downloads/pmtable.py", line 61, in <module>
    print(f"{i:04X}: {struct.unpack('f', os.read(devmem, 4))[0]}")
PermissionError: [Errno 1] Operation not permitted
[gamer@chimeraos Downloads]$ 
`

patrickschur, 2022-09-15:
@ruineka `CONFIG_IO_STRICT_DEVMEM` is not set in my config. Can you please disable it and try it again? If that's also not working, I would need the table base address and the output of `sudo cat /proc/iomem`. It might be possible that the command to get the base address is different on Rembrandt.

hh1599, 2022-09-19:
> @ruineka `CONFIG_IO_STRICT_DEVMEM` is not set in my config. Can you please disable it and try it again? If that's also not working, I would need the table base address and the output of `sudo cat /proc/iomem`. It might be possible that the command to get the base address is different on Rembrandt.

I compiled a kernel with strict_devmem=n and this is the result of `sudo python3 pmtable.py`

`table base address: 0xfe300000
table version: 0x450004
transfer table: []
0000: nan
0004: nan
0008: nan
000C: nan
0010: nan
0014: nan
0018: nan
001C: nan
0020: nan
0024: nan
0028: nan
002C: nan
0030: nan
0034: nan
0038: nan
003C: nan
0040: nan
0044: nan
0048: nan
004C: nan
0050: nan
0054: nan
0058: nan
005C: nan
0060: nan
0064: nan
0068: nan
006C: nan
0070: nan
0074: nan
0078: nan
007C: nan
0080: nan
0084: nan
0088: nan
008C: nan
0090: nan
0094: nan
0098: nan
009C: nan
00A0: nan
00A4: nan
00A8: nan
00AC: nan
00B0: nan
00B4: nan
00B8: nan
00BC: nan
00C0: nan
00C4: nan
00C8: nan
00CC: nan
00D0: nan
00D4: nan
00D8: nan
00DC: nan
00E0: nan
00E4: nan
00E8: nan
00EC: nan
00F0: nan
00F4: nan
00F8: nan
[... 594 lines cut ...]
0A44: nan
0A48: nan
0A4C: nan
0A50: nan
0A54: nan
0A58: nan
0A5C: nan
0A60: nan
0A64: nan
0A68: nan
0A6C: nan
0A70: nan
0A74: nan
0A78: nan
0A7C: nan
0A80: nan
0A84: nan
0A88: nan
0A8C: nan
0A90: nan
0A94: nan
0A98: nan
0A9C: nan
0AA0: nan
`

hh1599, 2022-09-19:
this is the result of `sudo cat /proc/iomem`

00000000-00000fff : Reserved
00001000-0009ffff : System RAM
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
00000000-00000000 : PCI Bus 0000:00
000a0000-000effff : PCI Bus 0000:00
000f0000-000fffff : System ROM
00100000-09bfffff : System RAM
09c00000-09dc0fff : Reserved
09dc1000-09efffff : System RAM
09f00000-09f27fff : ACPI Non-volatile Storage
09f28000-45014fff : System RAM
45015000-45123fff : Reserved
45124000-47370fff : System RAM
47371000-47371fff : Reserved
47372000-4f778fff : System RAM
4f779000-50ee8fff : Unknown E820 type
50ee9000-6c11cfff : Reserved
  6c101000-6c104fff : MSFT0101:00
    6c101000-6c104fff : MSFT0101:00
  6c105000-6c108fff : MSFT0101:00
    6c105000-6c108fff : MSFT0101:00
6c11d000-7ab7cfff : ACPI Non-volatile Storage
  7ab7b4ca-7ab7c4c9 : USBC000:00
7ab7d000-7abfcfff : ACPI Tables
7abfd000-7abfefff : System RAM
7abff000-7abfffff : Reserved
7ac00000-7b197fff : System RAM
7b198000-7b1a7fff : Reserved
7b1a8000-7bffcfff : System RAM
7bffd000-7cffffff : Reserved
7d800000-7fffffff : Reserved
80000000-dfffffff : PCI Bus 0000:00
  80000000-97ffffff : PCI Bus 0000:33
  98000000-afffffff : PCI Bus 0000:03
  b0000000-b03fffff : PCI Bus 0000:64
    b0000000-b00fffff : 0000:64:00.0
      b0000000-b00fffff : xhci-hcd
    b0100000-b01fffff : 0000:64:00.3
      b0100000-b01fffff : xhci-hcd
    b0200000-b02fffff : 0000:64:00.4
      b0200000-b02fffff : xhci-hcd
    b0300000-b037ffff : 0000:64:00.5
      b0300000-b037ffff : thunderbolt
    b0380000-b03fffff : 0000:64:00.6
      b0380000-b03fffff : thunderbolt
  b0400000-b07fffff : PCI Bus 0000:63
    b0400000-b04fffff : 0000:63:00.3
      b0400000-b04fffff : xhci-hcd
    b0500000-b05fffff : 0000:63:00.4
      b0500000-b05fffff : xhci-hcd
    b0600000-b06fffff : 0000:63:00.2
      b0600000-b06fffff : ccp
    b0700000-b077ffff : 0000:63:00.0
    b0780000-b07bffff : 0000:63:00.5
      b0780000-b07bffff : AMD ACP3x audio
        b0780000-b0790200 : acp_iomem
    b07c0000-b07c7fff : 0000:63:00.6
      b07c0000-b07c7fff : ICH HD audio
    b07c8000-b07cbfff : 0000:63:00.1
      b07c8000-b07cbfff : ICH HD audio
    b07cc000-b07cdfff : 0000:63:00.2
      b07cc000-b07cdfff : ccp
  b0800000-b08fffff : PCI Bus 0000:02
    b0800000-b0803fff : 0000:02:00.0
      b0800000-b0803fff : nvme
    b0804000-b0804fff : 0000:02:00.0
      b0804000-b0804fff : nvme
    b0805000-b0805fff : 0000:02:00.0
      b0805000-b0805fff : nvme
  b0900000-b09fffff : PCI Bus 0000:01
    b0900000-b0907fff : 0000:01:00.0
e0000000-efffffff : PCI MMCONFIG 0000 [bus 00-ff]
  e0000000-efffffff : Reserved
    e0000000-efffffff : pnp 00:00
f0000000-fdbfffff : PCI Bus 0000:00
fde00000-fdefffff : Reserved
  fde10510-fde1053f : MSFT0101:00
fe000000-fe07ffff : amd_iommu
feb00000-feb00007 : SB800 TCO
fec00000-fec003ff : IOAPIC 0
fec01000-fec013ff : IOAPIC 1
fec10000-fec1001f : pnp 00:03
fed00000-fed003ff : HPET 2
  fed00000-fed003ff : PNP0103:00
    fed00000-fed003ff : pnp 00:03
fed61000-fed613ff : pnp 00:03
fed80000-fed80fff : Reserved
  fed80000-fed80fff : pnp 00:03
fed81500-fed818ff : AMDI0030:00
fedc2000-fedc2fff : AMDI0010:00
  fedc2000-fedc2fff : AMDI0010:00 AMDI0010:00
fedc5000-fedc5fff : AMDI0010:03
  fedc5000-fedc5fff : AMDI0010:03 AMDI0010:03
fee00000-fee00fff : Local APIC
  fee00000-fee00fff : pnp 00:00
100000000-3fe2fffff : System RAM
  1dca00000-1dda01e4b : Kernel code
  1ddc00000-1de24afff : Kernel rodata
  1de400000-1de60acff : Kernel data
  1decab000-1deffffff : Kernel bss
3fe300000-3ff33ffff : RAM buffer
3ff340000-ffffffffff : Reserved
  480000000-ffffffffff : PCI Bus 0000:00
    4b0000000-4d7ffffff : PCI Bus 0000:33
    4e0000000-507ffffff : PCI Bus 0000:03
    510000000-5201fffff : PCI Bus 0000:63
      510000000-51fffffff : 0000:63:00.0
      520000000-5201fffff : 0000:63:00.0
    520200000-5202fffff : PCI Bus 0000:01
      520200000-5202fffff : 0000:01:00.0
        520200000-5202fffff : 0000:01:00.0
3fff80000000-3fffffffffff : 0000:63:00.0

GreyXor, 2022-09-20:
I have a 6850U running Arch Linux, how can I help with support for this processor?

hh1599, 2022-09-20:
@GreyXor As of right now, the instructions have been to run pmtable.py (https://github.com/FlyGoat/RyzenAdj/issues/210#issuecomment-1239900731) and `sudo cat /proc/iomem`.

you might have to compile a kernel with devmem support (devmem=y, strict_devmem=n is what i did).

hh1599, 2022-09-20:
Devs, 
I have full devmem access and AMDuProf working on Arch with a 6800U.  If there are any scripts or AMDuProfCLI commands you want me to run please let me know.

ruineka, 2022-09-20:
> @ruineka 
> ?
> I did that above. https://github.com/FlyGoat/RyzenAdj/issues/210#issuecomment-1251257558

Oh sorry I should have read the name more closely 😂

JadedTuna, 2022-09-21:
@patrickschur I ran the script two times, both yielded the same output.
That output and a dump of `iomem` can be found here: https://gist.github.com/JadedTuna/9a75f2c327da4cda116ab7e464ab2067

hh1599, 2022-09-21:
@JadedTuna any insight as so why my pmtable just outputs "nan"?  is it a 6800U vs 6850U thing or did I screw something up?

I definitely have devmem access as indicated by running this command: `sudo dd if=/dev/mem of=/dev/null bs=1 skip=1048576 count=1`

GreyXor, 2022-09-22:
https://gist.github.com/GreyXor/4424925d61e0a519c8af10e02efc6720

JadedTuna, 2022-09-23:
> @JadedTuna any insight as so why my pmtable just outputs "nan"? is it a 6800U vs 6850U thing or did I screw something up?
> 
> I definitely have devmem access as indicated by running this command: `sudo dd if=/dev/mem of=/dev/null bs=1 skip=1048576 count=1`

Good question. I am running an unmodified Fedora 36 KDE spin, for me things worked out of the box. FWIW my hardware is same as that of @GreyXor

patrickschur, 2022-09-24:
Thanks to everyone who provided the script output. Unfortunately the command to get the table base address is not correct. This is also the reason why everyone got a different result. 

The commands were reverse engineered from Ryzen Master. But it looks like AMD changed the commands or they are only valid for engineering samples. So the next step is to find the command that returns the base address of the PM table. You can use the `smn_send_msg` function from the script for that. Just pass the command ID to the function and test the first 192 commands (should be enough) and then send us the output. Don't forget to log the output, because some commands will freeze your system or the SMU will reject every command. In this case, you must restart your system and skip the command in the next run.

If you want, I can also provide you a script that does it for you.

GreyXor, 2022-09-24:
> 
Please yes, i'll execute it asap

hh1599, 2022-09-24:
@patrickschur thank you, I appreciate your time on this. 

I'm no programmer, but I'm no dummy either. If you can provide a sample script to send the commands I think I can figure out how to skip the commands that lock the system and log the output.

patrickschur, 2022-09-24:
@hh1599 and @GreyXor: Thanks for your help. Here is the script: https://gist.github.com/patrickschur/b0feaaea79595c4d632bc7fffb2774c6

The output should look like this (the first number is the command ID in hex):
```
01: ['0x1', '0x0', '0x0', '0x0', '0x0']
02: ['0x402d00', '0x0', '0x0', '0x0', '0x0']
```

The script will run until it reaches the command ID `bf`. If the script fails or runs in an endless loop, you must add the failed command ID to the list in [line 13](https://gist.github.com/patrickschur/b0feaaea79595c4d632bc7fffb2774c6#file-scan-py-L13). Then restart your system. The next time you run the script it will skip the failed command ID.

In my **example** from above the failed command ID would be `03` (last command ID + 1). So I would have to add `0x03` to the list.

GreyXor, 2022-09-24:
```python
SKIP_COMMANDS = [0x7b, 0x50, 0x60, 0x3c, 0x65, 0x76, 0x5d, 0x82, 0x40, 0x5e, 0x5d, 0x61, 0x7b, 0x7f, 0x3a]
```
Is it normal that I already have so many?
and sometimes the script blinks and there are two different lists displayed one after the other. which one should I use to enhance my skiplist?

patrickschur, 2022-09-24:
@GreyXor What do you mean with it blinks and there are two different lists displayed? And yes, it's normal to have so many. Can you already send me the results. Maybe you already found the command I was looking for. :)

GreyXor, 2022-09-24:
idk how to save the result, even with redirecting like `python scan.py > output.txt 2>&1` output is empty

patrickschur, 2022-09-24:
`sudo python3 scan.py > output.txt` should work totally fine. I guess the SMU locks your system. As I said, I probably don't need all 192 commands. Feel free to reduce the number from 192 to something like 96 (where it doesn't fail anymore). https://gist.github.com/patrickschur/b0feaaea79595c4d632bc7fffb2774c6#file-scan-py-L46

GreyXor, 2022-09-25:
I already run my cmd as root, i can't have anything in output.txt. and even when i reduce the number to 96 there is lot of fail

GreyXor, 2022-09-25:
https://gist.github.com/GreyXor/243ef35914a530488f0f9030d368077b

patrickschur, 2022-09-25:
@GreyXor Thanks! Unfortunately the output doesn't include the command I was looking for. Can you please proceed and skip the first 60 commands?
```py
for i in range(0x60, 0xC0):
```

GreyXor, 2022-09-25:
I run it, my pc freezes and output.txt is empty

GreyXor, 2022-09-25:
the output file remains empty. I even tried doing this 
```python
f = open('output.txt', 'w')

for i in range(0x60, 0xC0):
    if i not in SKIP_COMMANDS:
        r = [hex(v) for v in smn_send_msg(i, 0, 0)]
        f.write(f'{i:02x}: {r}\n')
        time.sleep(0.1)

os.close(config)
```
If I stop the execution before the crash then output.txt is filled, but when there is a crash, output.txt is empty on restart. very strange

GreyXor, 2022-09-25:
```bash
62: ['0x0', '0x0', '0x0', '0x0', '0x0']
63: ['0x0', '0x0', '0x0', '0x0', '0x0']
64: ['0x0', '0x0', '0x0', '0x0', '0x0']
66: ['0x7e300000', '0x6', '0x0', '0x0', '0x0']
67: ['0xfd']
68: ['0xfd']
69: ['0xfd']
6a: ['0xfd']
6b: ['0xfd']
6d: ['0xfd']
6e: ['0xfd']
6f: ['0xfd']
70: ['0xfd']
71: ['0xfd']
72: ['0xfd']
73: ['0xfd']
74: ['0xfd']
75: ['0xfd']
77: ['0xfd']
78: ['0xfd']
79: ['0xfd']
7a: ['0xfd']
7c: ['0xff']
7d: ['0x5', '0x0', '0x0', '0x0', '0x0']
7e: ['0x0', '0x0', '0x0', '0x0', '0x0']
80: ['0xfd']
81: ['0xfd']
83: ['0x0', '0x0', '0x0', '0x0', '0x0']
84: ['0x0', '0x0', '0x0', '0x0', '0x0']
85: ['0xfd']
86: ['0x0', '0x0', '0x0', '0x0', '0x0']
87: ['0x0', '0x0', '0x0', '0x0', '0x0']
88: ['0x0', '0x0', '0x0', '0x0', '0x0']
89: ['0x0', '0x0', '0x0', '0x0', '0x0']
8a: ['0x0', '0x0', '0x0', '0x0', '0x0']
8b: ['0xfd']
8c: ['0xfd']
8d: ['0xfd']
8e: ['0xfd']
8f: ['0x0', '0x0', '0x0', '0x0', '0x0']
90: ['0x0', '0x0', '0x0', '0x0', '0x0']
91: ['0x0', '0x0', '0x0', '0x0', '0x0']
92: ['0x0', '0x0', '0x0', '0x0', '0x0']
93: ['0x0', '0x0', '0x0', '0x0', '0x0']
94: ['0x0', '0x0', '0x0', '0x0', '0x0']
95: ['0x0', '0x0', '0x0', '0x0', '0x0']
96: ['0x0', '0x0', '0x0', '0x0', '0x0']
97: ['0x0', '0x0', '0x0', '0x0', '0x0']
98: ['0x0', '0x0', '0x0', '0x0', '0x0']
99: ['0x0', '0x0', '0x0', '0x0', '0x0']
9a: ['0x0', '0x0', '0x0', '0x0', '0x0']
9b: ['0x0', '0x0', '0x0', '0x0', '0x0']
9c: ['0x0', '0x0', '0x0', '0x0', '0x0']
9d: ['0x0', '0x0', '0x0', '0x0', '0x0']
9e: ['0x0', '0x0', '0x0', '0x0', '0x0']
9f: ['0x0', '0x0', '0x0', '0x0', '0x0']
a0: ['0x0', '0x0', '0x0', '0x0', '0x0']
a1: ['0x0', '0x0', '0x0', '0x0', '0x0']
a2: ['0x0', '0x0', '0x0', '0x0', '0x0']
a3: ['0x0', '0x0', '0x0', '0x0', '0x0']
a4: ['0x0', '0x0', '0x0', '0x0', '0x0']
a5: ['0x0', '0x0', '0x0', '0x0', '0x0']
a6: ['0x0', '0x0', '0x0', '0x0', '0x0']
a7: ['0x0', '0x0', '0x0', '0x0', '0x0']
a8: ['0x0', '0x0', '0x0', '0x0', '0x0']
a9: ['0x0', '0x0', '0x0', '0x0', '0x0']
aa: ['0x0', '0x0', '0x0', '0x0', '0x0']
ab: ['0x2980', '0x0', '0x0', '0x0', '0x0']
ac: ['0x1c60', '0x0', '0x0', '0x0', '0x0']
ad: ['0x0', '0x0', '0x0', '0x0', '0x0']
ae: ['0x23f', '0x0', '0x0', '0x0', '0x0']
af: ['0x230', '0x0', '0x0', '0x0', '0x0']
b0: ['0xfd']
b1: ['0xfd']
b2: ['0x0', '0x0', '0x0', '0x0', '0x0']
b3: ['0x0', '0x0', '0x0', '0x0', '0x0']
b4: ['0xfe']
b5: ['0xfe']
b6: ['0xfd']
b7: ['0xff']
b9: ['0x0', '0x0', '0x0', '0x0', '0x0']
ba: ['0x2cb', '0x0', '0x0', '0x0', '0x0']
bb: ['0xfd']
bc: ['0xfd']
bd: ['0xfd']
be: ['0xff']
bf: ['0xfd']
```

hh1599, 2022-09-25:
@patrickschur do you think there is a difference, for this purpose, between the 6800u and the 6850u? I'm on the road for work but I can run the script on my laptop as well when I get home tomorrow.

hh1599, 2022-09-27:
@patrickschur Here are my results for the 6800U: https://gist.github.com/hh1599/d706ca25a8b0cd6a4b77b213e983c725

My skip commands were: 0x5d,0x61,0x97,0x99

As an aside, for anyone else who wants to run the script.  I had a heck of a time getting terminal to output to a text file.  I think it was buffering the output and the PC would crash before it could write the file.  I modified the script to write to the file itself on every iteration and it worked great.  https://gist.github.com/hh1599/c8d6c7ffca6089eb9498e4cbace2f3a2

I also ran into a problem where entire blocks of skip commands still wouldn't allow the script to progress.  But when i would change the script to start just before that block (`for i in range(100, 192)`): it would run through them all fine.  I think earlier commands were interfering with later ones.

If there is anything else you want me to do let me know.

hh1599, 2022-10-02:
@patrickschur I know your probably super busy, and I appreciate all the time you have spent helping us, but I have to return this laptop (thinkpad 13s, stock nvme died on day 1 and the keyboard doesnt work on linux).  Is there anything else I can run for you before I do?

Thanks.

patrickschur, 2022-10-03:
@hh1599 Sorry for the late reply. I'm indeed busy and already working on support for Raphael.

> Is there anything else I can run for you before I do?
No, not at the moment. Thanks for your support.

GreyXor, 2022-10-04:
@patrickschur Do you have what you want from my last output?
Thank you

patrickschur, 2022-10-10:
@GreyXor Yes, I have everything I wanted. Thanks! Unfortunately, I was not able to find the command to get the table base address yet. Based on the output you send me I would guess that `0x66` is the command we are looking for. The value (`0x7e300000`) looks like an address and it's also part of the reserved memory region in `/proc/iomem`. Which is a good sign.

```
7077e000-7697dfff : Reserved
```

But that's the same command I already used in my first script. So either the command is wrong or the command to update the table... If you have some time you can poke some commands around `0x66` to find the update table command. Both commands are often close together (but don't have to). Just use the first script I send you and replace `SMN_CMD_TRANSFER_TABLE` with the command ID you want to test or do something like this (replace line 55):

```py
# 0x65 was not part of your output, so it might fail... Lets try it anyway.
for command_id in [0x62, 0x63, 0x64, 0x65, 0x67, 0x68, 0x69, 0x6a, 0x6b]:
    for table_id in range(10):
        smn_send_msg(command_id, table_id)
```

Hopefully one of the commands will transfer the table...

runt9, 2022-11-01:
Just got an ASUS Vivobook 15x with a 6900HX and immediately loaded Ubuntu on it, tried to restore my work database, only to be met by thermal cutoff. Unfortunately the ASUS BIOS does not have CPU settings as you're supposed to use the MyASUS app in Windows.

So I'm here looking for RyzenAdj support for the 6xxx series so I can tune down the proc a bit, looks like there's some scripts to run but I'm not 100% sure what the most up-to-date script is. I'm happy to help out any way I can as right now my laptop isn't particularly usable with thermal cutoff being so easy to hit.

Let me know what you'd like me to run and I'll get on it ASAP.

ruineka, 2022-11-01:
> Just got an ASUS Vivobook 15x with a 6900HX and immediately loaded Ubuntu on it, tried to restore my work database, only to be met by thermal cutoff. Unfortunately the ASUS BIOS does not have CPU [...]

Use the 6.1 kernel RC. It has the Asus ACPI fix that covers our laptops. With my Zephyrus my fan wouldn't kick on properly or not at all.

runt9, 2022-11-01:
> Use the 6.1 kernel RC. It has the Asus ACPI fix that covers our laptops. With my Zephyrus my fan wouldn't kick on properly or not at all.

Oof, part of the reason I dodged getting a 12th gen Intel was I wanted to stick to Ubuntu 22.04 which is on 5.15 and the 12th gen Intel scheduler requires 5.18. Jumping to an RC kernel is a whole other world. I assume since it's still in RC, it's in full "pull code and compile yourself" territory? Which has the problem of I'm not 100% sure what all features and flags I'd need to turn on in order to match what I'm currently running safely.

ruineka, 2022-11-01:
> Oof, part of the reason I dodged getting a 12th gen Intel was I wanted to stick to Ubuntu 22.04 which is on 5.15 and the 12th gen Intel scheduler requires 5.18. Jumping to an RC kernel is a whole [...]

Ubuntu has a nice tool called 'Ubuntu Mainline " that allows you to update the kernel. Go to settings to enable RC kernels and RC2 or 3 should be there. It's been stable for me so far.

chakravala, 2022-11-09:
> But that's the same command I already used in my first script. So either the command is wrong or the command to update the table... If you have some time you can poke some commands around `0x66` to find the update table command. Both commands are often close together (but don't have to). Just use the first script I send you and replace `SMN_CMD_TRANSFER_TABLE` with the command ID you want to test or do something like this (replace line [55](https://gist.github.com/patrickschur/4645f81601fe6f8ff16c7a90e848d7cb#file-pmtable-py-L55)):
> 
> ```python
> # 0x65 was not part of your output, so it might fail... Lets try it anyway.
> for command_id in [0x62, 0x63, 0x64, 0x65, 0x67, 0x68, 0x69, 0x6a, 0x6b]:
>     for table_id in range(10):
>         smn_send_msg(command_id, table_id)
> ```
> 
> Hopefully one of the commands will transfer the table...

Tried the snippet here and got this result: (haven't recompiled my kernel though)
```
table base address: 0x3e300000
table version: 0x450005
transfer table: [0, 6, 0, 0, 0]
```
hope that helps, I got 6800U and want to help out

hh1599, 2022-11-20:
> @GreyXor Yes, I have everything I wanted. Thanks! Unfortunately, I was not able to find the command to get the table base address yet. Based on the output you send me I would guess that `0x66` is the command we are looking for. The value (`0x7e300000`) looks like an address and it's also part of the reserved memory region in `/proc/iomem`. Which is a good sign.
> 
> ```
> 7077e000-7697dfff : Reserved
> ```
> 
> But that's the same command I already used in my first script. So either the command is wrong or the command to update the table... If you have some time you can poke some commands around `0x66` to find the update table command. Both commands are often close together (but don't have to). Just use the first script I send you and replace `SMN_CMD_TRANSFER_TABLE` with the command ID you want to test or do something like this (replace line [55](https://gist.github.com/patrickschur/4645f81601fe6f8ff16c7a90e848d7cb#file-pmtable-py-L55)):
> 
> ```python
> # 0x65 was not part of your output, so it might fail... Lets try it anyway.
> for command_id in [0x62, 0x63, 0x64, 0x65, 0x67, 0x68, 0x69, 0x6a, 0x6b]:
>     for table_id in range(10):
>         smn_send_msg(command_id, table_id)
> ```
> 
> Hopefully one of the commands will transfer the table...

@patrickschur So I dont really know what im doing but i modified your original pmtable.py with the sample code you provided [here](https://gist.github.com/hh1599/fdc6cdeffe3583c6b7e94a279cfda17b#file-pmtable-py) and got these results [here](https://gist.github.com/hh1599/fdc6cdeffe3583c6b7e94a279cfda17b#file-results).

Tapchicoma, 2022-11-20:
Using the same script as @hh1599 here is [my dump](https://gist.github.com/Tapchicoma/777079155b218983161f17d3c11f35c4) for 6950H.

hh1599, 2022-11-20:
@Tapchicoma

If you check the link again i modified the script to include more commands since it didnt seem like there was any useful information in there.

@patrickschur

patrickschur, 2022-11-25:
Sorry for the late answer but I'm very busy at the moment. Nevertheless I made some changes. Could someone please test the script? I might be possible you have to execute it twice to see a difference. https://gist.github.com/patrickschur/4645f81601fe6f8ff16c7a90e848d7cb

Tapchicoma, 2022-11-25:
Script fails in my case with following error:
```
base address (high): 0x8, (low): 0x5e300000
table version: 0x450005
table base address: 0x85e300000
transfer table: [4521989, 8, 0, 0, 0]
Traceback (most recent call last):
  File "pmtable.py", line 64, in <module>
    print(f"{i:04X}: {struct.unpack('f', os.read(devmem, 4))[0]}")
PermissionError: [Errno 1] Operation not permitted
```

patrickschur, 2022-11-25:
@Tapchicoma Please make sure your kernel is compiled with `/dev/mem` support. The topic was already discussed in this issue. See https://github.com/FlyGoat/RyzenAdj/issues/210#issuecomment-1247406266.

[44 more replies]

closed 2023-06-27 by danyi