FlyGoat/RyzenAdj #218: Restore all parameters to default

k98p, 2022-09-07:
I used the following command to set the max temperature at 90 degrees Celcius:
`./ryzenadj --tctl-temp=90`

However, whenever I restart my system, a warning comes from the Riot anti-cheat system that something has been blocked. This became regular after the above operation. Could you please guide me through the steps to uninstall or set the parameter back to its default value?

[image]

 Thank you!

Falcosc, 2022-09-15:
The value get reset on reboot because anything which is done with ryzenadj is only temporary.

You need to make sure that nothing is calling ryzenadj. If you are uncertain if something is calling ryzenadj, then you can also just delete it.

If you are certain that nothing is calling ryzenadj, then this error is not related to ryzenadj.

https://github.com/FlyGoat/RyzenAdj/wiki/FAQ#inpoutx64-got-blocked-by-anti-cheat-software

pcislocked, 2023-03-15:
its probably something else, no vanguard error on my computer.