FlyGoat/RyzenAdj #219: No error but no tdp limit change

SpatialZebra33, 2022-09-07:
Ryzen 7 3750H
HP Pavilion Gaming

Always limited to 15W, real problem for my gaming performance.
AMD specifies this processor as 35W capable, and has a cTDP, but ofc HP has no option in BIOS.

SpatialZebra33, 2022-09-07:
After messing with it, I noticed that my fan control app (blocking them at 100%) was interfering so I disabled it.
Still it doesn't work well.

I need to:
-Be on battery
-reboot
-Execute ryzenadj (every tdp set to 35W and temp limit to 90)
-Do something resource intensive on battery (such as Cinebench)
-It will draw 15W
-Suddently plug in the charger
-It will go over 15W, but won't really exceed 23
-Stop Cinebench and launch my game

It won't be stuck at 15W but tends to stay pretty low, usually around 10 or lower just like without ryzenadj, even when temperature DEFINITELY allows for higher power.
It will randomly spike over 20W for a very short period of time.

I specified long boost, short boost and normal TDP all at 35W...

Falcosc, 2022-09-15:
--info will show you the current value. If you don't see your 35W there, then you can not do anything.
If your value is only visible for a short period of time, then something else does overwrite it.

You can brute force it by setting the value over and over again to overwrite the overwrite from the unknown source. For windows users we have shared a script to automate overwriting of other configuration changes: https://github.com/FlyGoat/RyzenAdj#windows-installation

SpatialZebra33, 2022-09-16:
35W just doesn't happen (or for under 500 ms)
Given how badly background processes affect my Bedwars experience I wouldn't want a program to repeatedly change the max TDP, it'd probably make my FPS super unstable...
What would be interesting is locking the iGPU at a certain frequency, allowing for more power to the CPU as it's not rare to see it under 10W during gaming sessions.

Falcosc, 2022-09-16:
Combine your option with --info to find out if it gets applied at all.
iGPU freqzency adjustments are not supported on your cpu generation

SpatialZebra33, 2022-09-20:
It says RyzenAdj was applied successfully.
For the iGPU clock I don't need to decrease it I only wanted to do so for it to use less power, and thus let the CPU use more.

Falcosc, 2022-09-20:
Combine your option with --info in one command to find out if the value gets written at all.
The RyzenAdj was applied successfully is displayed if your system did not reject it. But accepting values does not mean that they are actually written. And ryzenAdj does not monitor which values got written, it's just handling the response of the command.

SpatialZebra33, 2022-09-26:
Well it even for a short period of time I don't see 35W at all.
But at this point as I learned that most of the Minecraft stutter weirdly enough comes from allocating more RAM (seems way more stable frame graph at 2GB), I don't think I could even do much by getting my TDP higher.
Also this laptop already had one of the RAM lanes partially dying, being rendered unusable, due to overheating.
Basically about a year ago I got a BSoD and this happened...

VaxYourDamnKids, 2022-10-29:
> 65W Liquid Metal, bruh. R5 3450U Vega 8 running @ 1.2k at allt imes. B-)

[image]