FlyGoat/RyzenAdj #226: Easy Anti Cheat, WinRing0 incompatibility

Valkirie, 2022-11-13:
As the title suggests, using Ryzenadj will trigger EAC (Easy Anti Cheat) due to its usage of WinRing0. I believe their isn't much we can do here. I tried several release and dlls I could find across the web but all of them triggered EAC.

Falcosc, 2022-11-13:
https://github.com/FlyGoat/RyzenAdj/wiki/FAQ#inpoutx64-got-blocked-by-anti-cheat-software

> inpoutx64.dll contains inpoutx64.sys which is a driver to enable access to the whole physical memory. It is used to access the PM Table to enable monitoring. You can remove the file if you have trouble with it, adjustments will still work. Only --info and --dump-table does use the PM Table. If you are using the PowerShell script, you need to disable monitorField because monitoring will not work without inpoutx64.dll

cronyakatsuki, 2023-01-13:
> https://github.com/FlyGoat/RyzenAdj/wiki/FAQ#inpoutx64-got-blocked-by-anti-cheat-software
> 
> > inpoutx64.dll contains inpoutx64.sys which is a driver to enable access to the whole physical memory. It is used to access the PM Table to enable monitoring. You can remove the file if you have trouble with it, adjustments will still work. Only --info and --dump-table does use the PM Table. If you are using the PowerShell script, you need to disable monitorField because monitoring will not work without inpoutx64.dll
> 
> If it does still not work, you have to avoid using ryzenAdj while in the Game. Your adjustments will stay if you have done them before starting the game. If avoid starting ryzenAdj doesn't help, you can write a script which removes the file while starting the game but restores it on demand.

It is the problem with having any of the service running or anything in the background that even touch WinRing0, not with the file itself.

Also some people, including me have laptops that will change tdp by themself sometimes especially while playing games so easy anticheat is quite stupid for not being able to just look at what the process is even doing before just saying "YOU CHEATING" and closing the game.

closed 2024-04-21 by Valkirie