FlyGoat/RyzenAdj #231: SecureBoot and PCI access - wish to dig in

OhSoGood, 2022-12-01:
Hi,
On both Ubuntu and Debian I get the error `pcilib: sysfs_write: write failed: Operation not permitted` that most of use faced. Disabling Secure Boot indeed solves it, but for many other reasons I'd like to both use RyzenAdj and Secure Boot. There was a beginning of discussion on that thread https://github.com/FlyGoat/RyzenAdj/issues/42 but it didn't go that far.

I wonder if any could lead me on the right direction to try to solve this (kernel compil option? apparmor? etc.). Any link, ideas, etc would be welcomed.

And maybe the most important ;-) Thank you for all the work and time around RyzenAdj (@FlyGoat & co) !

FlyGoat, 2022-12-03:
FYI: https://github.com/torvalds/linux/commit/eb627e17727ebeede70697ae1798688b0d328b54

atassis, 2023-04-27:
Having the same point of interest in this. Can we somehow sign ryzenadj with MOK keys to let it control the system? Like it does during nvidia drivers installation

Agno94, 2024-01-16:
Has anyone ever tried using https://github.com/blastrock/unlockdown as a workaround?

Agno94, 2024-01-19:
I tried the `unlockdown` module and **it works**. 

#### Description

Normally this should be the full workaround. I had to sign it manually but probably it's a problem with my setup (see below). But most on most distribution dkms should sign the module it builds.

1. clone the repository `git clone https://github.com/blastrock/unlockdown.git`
2. build and install `cd unlockdown; sudo make dkmsInstall`
3. load the module to disable kernel lockdown `sudo modproble unlockdown`
4. execute ryzenAdj `sudo ryzenadj -i`
5. unload the module to restore kernel lockdown `sudo modproble -r unlockdown`

In case like mine where dkms signing don't work you can sign the `.ko` file of the module with this command:

`sudo /usr/src/linux-headers-$YOUR_KERNEL_VERSION/scripts/sign-file sha256 $PATH_OF_YOUR_SECURE_BOOT_MOK_OR_DB_PRIVATE_KEY $PATH_OF_YOUR_SECURE_BOOT_MOK_OR_DB_PUBLIC_KEY_CERT $MODULE_DIRECTORY/unlockdown.ko `

In my case `MODULE_DIRECTORY=/lib/modules/$YOUR_KERNEL_VERSION/updates/dkms`

#### Questions

Do you think this is a valid workaround?

Turning off kernel lockdown as security implications but is surely more secure then turning off secureboot entirely for someone who wants to try out this software.

Can someone else make it work?

Saiyaken-PHOENIX, 2024-03-27:
> Can anyone else make it work?

@Agno94 
Just attempted the unlockdown dkmsInstall, don't think it works with Linux-zen kernel releases unfortunately, this is what I got. 

`sudo dkms remove -m unlockdown -v 0.01 --all || true
Error! The module/version combo: unlockdown-0.01 is not located in the DKMS tree.
sudo rm -fr /usr/src/unlockdown-0.01/
sudo mkdir -p /usr/src/unlockdown-0.01/
sudo cp Kbuild dkms.conf Makefile mod.c /usr/src/unlockdown-0.01/
sudo dkms add -m unlockdown -v 0.01
Creating symlink /var/lib/dkms/unlockdown/0.01/source -> /usr/src/unlockdown-0.01
sudo dkms install unlockdown/0.01
Error! Your kernel headers for kernel 6.8.1-zen1-1-zen cannot be found at /usr/lib/modules/6.8.1-zen1-1-zen/build or /usr/lib/modules/6.8.1-zen1-1-zen/source.
Please install the linux-headers-6.8.1-zen1-1-zen package or use the --kernelsourcedir option to tell DKMS where it's located.
make: *** [Makefile:41: dkmsInstall] Error 1
`

Agno94, 2024-03-27:
> Your kernel headers for kernel 6.8.1-zen1-1-zen cannot be found at /usr/lib/modules/6.8.1-zen1-1-zen/build or /usr/lib/modules/6.8.1-zen1-1-zen/source

Are you using Arch? From what I see the linux-zen kernel headers should be provided by [this package](https://archlinux.org/packages/extra/x86_64/linux-zen-headers/) and saved in `/usr/lib/modules/6.8.1-zen1-1-zen/build/`.

Eckoa, 2024-06-06:
Im using Arch and lockdown doesnt seem to be enabled by default on the default kernel when secure boot is enabled. Is there anyone else able to confirm lockdown status on Arch? 

I have secure boot with custom keys and uki setup and working with ryzenadj no problem, obviously without lockdown there is some security risk but at the very least the Arch devs seem to make it optional.

hariganti, 2025-04-01:
I think it'd be helpful to mention to check kernel lockdown mode regarding the `pcilib: sysfs_write: write failed: Operation not permitted` error since I tried disabling Secure Boot with no success, then stumbled across this. It doesn't even seem that Secure Boot it, itself, the issue

AlxHnr, 2025-05-08:
I've implemented a feature, which allows using RyzenAdj on Linux without disabling secure boot and without disabling lockdown mode. It uses the ryzen_smu module instead of `/dev/mem`. Those who don't want to wait until my changes get reviewed and merged can already use this guide I've wrote: https://gist.github.com/AlxHnr/2e5cae579fba7a7e9f8358b60f025bab

closed 2025-05-18 by FlyGoat

kevlar700, 2026-09-04:
I hope this and the module gets upstreamed or the kernel gets this functionality. Without --tctl-temp=80 then the cpus surrounding components are stressed beyond their rated temperature

PR #341 (2025-05-18): Add full support for ryzen_smu Linux kernel module, making `/dev/mem` and libpci optional
RyzenAdj has partial support for the ryzen_smu kernel module, but uses it only for optional retrieval of the PM table. With this merge request I'm adding a new CMake feature flag, which allows using RyzenAdj on Linux **without** exposing `/dev/mem` and **without** needing libpci.

Closes #231