FlyGoat/RyzenAdj #233: 6850U: set_coall is rejected by SMU

nicolasf, 2022-12-11:
"set-coall" is not working on a 6850U.

I saw that there is code specific for Rembrandt, but I'm unsure if it works with other Zen3+ CPUs and mine specifically won't accept it.

Result with DEBUG:
```
sudo ./ryzenadj --set-coall=0x100000
SMU_SERVICE REQ_ID:0x4c
SMU_SERVICE REQ: arg0: 0x100000, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x100000, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
set_coall is rejected by SMU
```

Code supposedly specific to Rembrandt:
```
EXP int CALL set_coall(ryzen_access ry, uint32_t value) {
    int err = ADJ_ERR_FAM_UNSUPPORTED;

	switch (ry->family)
	{
	case FAM_CEZANNE:
	case FAM_RENOIR:
	case FAM_LUCIENNE:
		_do_adjust(0x55);
		break;
	case FAM_REMBRANDT:
		_do_adjust(0x4C);
		break;
	}
	return err;
}
```

Nagidrop, 2022-12-29:
I think the problem not lying in the RyzenAdj but due to the 6850U (and most AMD mobile CPUs) not supporting it. I have a 5800H and it's also rejected by SMU when running --set-coall or --set-coper. Only CPUs unlocked for overclocking are able to run such commands as far as I know (5900HX, 5900HS,...)

JamesCJ60, 2023-05-12:
Hi, from the 5000 series and newer Curve Optimiser on mobile APUs only works on Ryzen 9 models.

closed 2023-05-12 by JamesCJ60

mengt1415, 2023-10-30:
> I think the problem not lying in the RyzenAdj but due to the 6850U (and most AMD mobile CPUs) not supporting it. I have a 5800H and it's also rejected by SMU when running --set-coall or --set-coper. [...]
 hi, could you please tell how to use --set-coper ?