FlyGoat/RyzenAdj #234: Random request_transfer_table is unsupported/failed popping up

Nagidrop, 2022-12-17:
I've been using the [readjustService.ps1](https://github.com/FlyGoat/RyzenAdj/blob/master/win32/readjustService.ps1) for a while to apply the slow_limit, fast_limit and stapm_limit without problems. However, sometimes, there are random errors printed: 
&nbsp;&nbsp;&nbsp;&nbsp;"request_transfer_table is unsupported"
or "request_transfer_table failed with unknown response 65"

I did a quick search and found out it was from the _return_translated_smu_error function of [api.c](https://github.com/FlyGoat/RyzenAdj/blob/master/lib/api.c) file, but I have no idea what the code does. I only have the above 2 specific request_transfer_table errors and not the others. They also appears to occur randomly before or after fast_limit unexpectedly changes while the script running in background. 

Do you have any idea what that means?

Falcosc, 2022-12-18:
Can be ignored. Sometimes 2 tools talk over each other.

Falcosc, 2022-12-18:
How did you even see these messages? I run this script via task scheduler, so it doesn't bother me.

Nagidrop, 2022-12-19:
> Can be ignored. Sometimes 2 tools talk over each other. If that happens, you get garbage from the other tool in the response register and that confused the response message handling.

It's kinda weird because I have only RyzenAdj tool running which controls the CPU (I have LegionFanControl as well but it only controls the fan speed at temp thresholds), but I'm glad that it can safely ignored. 

> How did you even see these messages? I run this script via task scheduler, so it doesn't bother me.

I ran the script via PowerShell as Administrator because I don't need the TDP limit outside of games. I did modify the script here and there and clean up some redundant code, but these errors pop up even at the default code (with greater frequency as well).

closed 2022-12-20 by Falcosc