FlyGoat/RyzenAdj #235: 5650u - resetSTAPMusage fails

asymcon, 2022-12-20:
Hi,
I'm running Cezanne 5650u, pretty much any adjustments are locked. Exactly as README pointed out, the commands go through successfully, but none of the limits are changed. I tried my luck with resetSTAPMUsage, but that hasn't worked either:

[image]

Is there any other workaround to get higher slow-limit?

Thanks

Falcosc, 2022-12-20:
I don't know workarounds for https://github.com/FlyGoat/RyzenAdj/wiki/Supported-Models#custom-value-range-for-zen3

The whole point of the invention of this new Zen3 hard lock was removing all workarounds.
 You can only try stuff and monitor the result. Maybe you find a new workaround.

valpackett, 2023-01-14:
The warning seems rather bogus!! On my 5850U Thinkpad the workaround *does* seem to work despite the warning.

Testing Cinebench R15:

- performance slider set to perf, no ryzenadj: 1240cb
- running with `$monitorField = "stapm_limit"` but `$resetSTAPMUsage = $false`: 1480cb
- adding `$resetSTAPMUsage = $true`: 1547cb (now we're talking!)

The debug log repeats like:

```
[STAPM_RESET] stapm_value (24.99126) nearing stapm_limit (25), resetting...
stapm_limit value unexpectedly changed from 25 to 16.40907
[STAPM_RESET] stapm_value (21.0428) nearing stapm_limit (16.40907), resetting...
set fast_limit to 45000
set slow_limit to 40000
set stapm_limit to 25000
set stapm_time to 500
set tctl_temp to 90
set apu_skin_temp_limit to 69
[STAPM_RESET] stapm_value (24.98986) nearing stapm_limit (25), resetting...
stapm_limit value unexpectedly changed from 25 to 16.40907
[STAPM_RESET] stapm_value (20.96571) nearing stapm_limit (16.40907), resetting...
set fast_limit to 45000
set slow_limit to 40000
set stapm_limit to 25000
set stapm_time to 500
set tctl_temp to 90
set apu_skin_temp_limit to 69
```