FlyGoat/RyzenAdj #237: Cannot set slow-time

ci70, 2022-12-21:
Cannot set slow-time on Ryzen 7 5800H even though it says it "sucessfully set."

```
Sucessfully set stapm_limit to 40000
Sucessfully set fast_limit to 45000
Sucessfully set slow_limit to 45000
Sucessfully set slow_time to 80
Sucessfully set stapm_time to 200
Sucessfully set tctl_temp to 90

```
```
ryzenadj -i

CPU Family: Cezanne
SMU BIOS Interface Version: 18
Version: v0.12.0
PM Table Version: 400005
|        Name         |   Value   |      Paramter      |
|---------------------|-----------|--------------------|
| STAPM LIMIT         |    40.000 | stapm-limit        |
| STAPM VALUE         |    11.437 |                    |
| PPT LIMIT FAST      |    45.000 | fast-limit         |
| PPT VALUE FAST      |    11.555 |                    |
| PPT LIMIT SLOW      |    45.000 | slow-limit         |
| PPT VALUE SLOW      |    11.357 |                    |
| StapmTimeConst      |   200.000 | stapm-time         |
| SlowPPTTimeConst    |    30.000 | slow-time          |
| PPT LIMIT APU       |    54.000 | apu-slow-limit     |
| PPT VALUE APU       |    11.457 |                    |
| TDC LIMIT VDD       |    58.000 | vrm-current        |
| TDC VALUE VDD       |     4.657 |                    |
| TDC LIMIT SOC       |    15.000 | vrmsoc-current     |
| TDC VALUE SOC       |     2.211 |                    |
| EDC LIMIT VDD       |   110.000 | vrmmax-current     |
| EDC VALUE VDD       |    75.587 |                    |
| EDC LIMIT SOC       |    20.000 | vrmsocmax-current  |
| EDC VALUE SOC       |     0.000 |                    |
| THM LIMIT CORE      |    90.000 | tctl-temp          |
| THM VALUE CORE      |    49.629 |                    |
| STT LIMIT APU       |     0.000 | apu-skin-temp      |
| STT VALUE APU       |     0.000 |                    |
| STT LIMIT dGPU      |     0.000 | dgpu-skin-temp     |
| STT VALUE dGPU      |     0.000 |                    |
| CCLK Boost SETPOINT |    50.000 | power-saving /     |
| CCLK BUSY VALUE     |    36.703 | max-performance    |
```

Falcosc, 2022-12-22:
That's normal: https://github.com/FlyGoat/RyzenAdj/wiki/FAQ#temperature-or-power-limit-does-not-work

Can somebody confirm this on his rig? Then we can mark it as not supported here: https://github.com/FlyGoat/RyzenAdj/wiki/Supported-Models

Seams like we got at least one guy with a working slow time or the green checkmark is wrong.

skylukemartin, 2023-06-30:
I also have the 5800h. You can change the value but it looks like it's capped at 30 for some reason. Default on my machine is 5.