FlyGoat/RyzenAdj #239: set_min_vcn is rejected by SMU on 3400G

abbaskn, 2022-12-22:
Hi,

I am unable to change the vcn clock speed to 600 MHz and getting the error set_min_vcn is rejected by SMU.

Is ryzenadj no longer supported for this APU?

abbaskn, 2022-12-22:
I tried from v0.12.0 to v0.8.3

Falcosc, 2022-12-23:
not supportet https://github.com/FlyGoat/RyzenAdj/wiki/Supported-Models

abbaskn, 2022-12-23:
Actually it worked for me with the same APU few months back with v0.8.3. There is also a youtube video exactly replicating my issue and how ryzenadj fixes it.

As I made it run as a scheduled task I was not aware when it started fail. As I was again facing the issue, I noticed that it is not working now.

 https://www.google.com/url?sa=t&source=web&cd=&ved=2ahUKEwismIXYzpD8AhVm8DgGHW83Dq4QwqsBegQICBAF&url=https%3A%2F%2Fm.youtube.com%2Fwatch%3Fv%3D0-whM6bXY08&usg=AOvVaw1eGTreS2ymgJYXScghg-S5

FlyGoat, 2022-12-23:
Perhaps AMD banned this feature with a firmware update.

abbaskn, 2022-12-23:
Yep looks like it. I tried installing Win 10 on spare SSD and tried to run the ryzenadj. Got the same error message. Its ironic how AMD could block something that fixes my issues but cannot fix the actual issue.

Anyway, I guess you can mark Picasso APU as Not Supported I guess for VCN. I tried with Ryzen 2400G, ryzenadj works there. But not sure that system had latest Firmware.

Thanks for the good work and support! Cheers!

FlyGoat, 2023-01-01:
Closing as we can't do anything...

closed 2023-01-01 by FlyGoat