FlyGoat/RyzenAdj #241: Can't make it work with 6600h

ostrovlyan, 2023-01-19:
Hi there. My spec are 6600h with manjaro 6.1.1-1
this is what i get when i tried to launch:
`CPU Family: Rembrandt
SMU BIOS Interface Version: 18
Version: v0.12.0 
failed to get /sys/kernel/ryzen_smu_drv/pm_table: No such file or directory
failed to map /dev/mem: Operation not permitted
If you don't want to change your memory access policy, you need a kernel module for this task.
We do support usage of this kernel module https://gitlab.com/leogx9r/ryzen_smu
Unable to get memory access
Unable to init power metric table: -5, this does not affect adjustments because it is only needed for monitoring.`

I also tried to get ryzen_smu, but got this:
`sudo dmesg | grep ryzen_smu                                                                                             
[    2.270202] ryzen_smu: loading out-of-tree module taints kernel.
[    2.270229] ryzen_smu: module verification failed: signature and/or required key missing - tainting kernel`

FlyGoat, 2023-01-19:
Hmm, disable secure boot?

ostrovlyan, 2023-01-20:
Never been enabled.

NikhilSaini38, 2023-01-27:
Have you tried using `iomem=relaxed` kernel param?

9Morello, 2025-03-10:
@NikhilSaini38 @ostrovlyan Doing that made it work on my Tumbleweed system with a Ryzen AI 9 HX 370.

clort81, 2025-04-07:
> Have you tried using `iomem=relaxed` kernel param?

That fixed it for me on Thinkpad T485 with Ryzen  3500u and kernel 6.1 (devuan).