FlyGoat/RyzenAdj #245: inpoutx64 Blocked in Latest W11 Insider Dev Build

bagusnl, 2023-02-10:
As the title issue suggest, the power table is not accessible due to inpoutx64 driver blocked by Windows.

RyzenAdj output:
```
>ryzenadj -i 
CPU Family: Picasso
SMU BIOS Interface Version: 8
Version: v0.12.0
Could not open inpoutx64 driver
Unable to get memory access
Unable to init power metric table: -5, this does not affect adjustments because it is only needed for monitoring.
```

Event viewer:
[image]
[image]

System Information:
Windows 11 Pro Insider Dev 
Build 25295.rs_prerelease.230203-1346
Secure boot OFF
fTPM ON

Ryzen 3 3200U
Acer Aspire 3 A315-45-R69L

FlyGoat, 2023-02-11:
https://www.elevenforum.com/t/enable-or-disable-microsoft-vulnerable-driver-blocklist-in-windows-11.10031/

bagusnl, 2023-02-11:
> https://www.elevenforum.com/t/enable-or-disable-microsoft-vulnerable-driver-blocklist-in-windows-11.10031/

Already have that in but it still can't load the driver
Adding it as exceptions on Windows Defender doesn't seem to help as well. Currently trying to do regular sfc and dism to see if my upgrade is somehow faulty.