FlyGoat/RyzenAdj #246: Ryzen 7040 APU series support (Phoenix)

GreyXor, 2023-03-01:
Few or no laptops are currently available with Phoenix 7040. This topic is here in a preliminary way.

smacz42, 2023-05-06:
I have a testing machine if needed.

ruineka, 2023-06-14:
I am unable to read tables on the Asus Ally with the Z1 Extreme which I think is considered a Phoenix series.
```
Unable to init power metric table: -1, this does not affect adjustments because it is only needed for monitoring.
```

patrickschur, 2023-06-16:
I'll take a closer look at Phoenix once my Framework mainboard (7840U version) arrives. But this might take 1-2 months.

FlyGoat, 2023-06-18:
I just got my ROG Zephyrus G14 with 7940HS.
Will try to spare some time to mess around :-)

thomas-ah, 2023-08-17:
On a GPW Win Max 2 2023 with a Ryzen 7840U running Arch Linux (kernel 6.5.0-rc6-1-mainline) I seem to get correct PM table output when running `ryzenadj -i`:

```
CPU Family: Phoenix
SMU BIOS Interface Version: 12
Version: v0.13.0
PM Table Version: 4c0007
|        Name         |   Value   |     Parameter      |
|---------------------|-----------|--------------------|
| STAPM LIMIT         |    22.000 | stapm-limit        |
| STAPM VALUE         |     4.361 |                    |
| PPT LIMIT FAST      |    24.000 | fast-limit         |
| PPT VALUE FAST      |    14.703 |                    |
| PPT LIMIT SLOW      |    24.000 | slow-limit         |
| PPT VALUE SLOW      |     4.418 |                    |
| StapmTimeConst      |       nan | stapm-time         |
| SlowPPTTimeConst    |       nan | slow-time          |
| PPT LIMIT APU       |       nan | apu-slow-limit     |
| PPT VALUE APU       |       nan |                    |
| TDC LIMIT VDD       |       nan | vrm-current        |
| TDC VALUE VDD       |       nan |                    |
| TDC LIMIT SOC       |       nan | vrmsoc-current     |
| TDC VALUE SOC       |       nan |                    |
| EDC LIMIT VDD       |       nan | vrmmax-current     |
| EDC VALUE VDD       |       nan |                    |
| EDC LIMIT SOC       |       nan | vrmsocmax-current  |
| EDC VALUE SOC       |       nan |                    |
| THM LIMIT CORE      |       nan | tctl-temp          |
| THM VALUE CORE      |       nan |                    |
| STT LIMIT APU       |       nan | apu-skin-temp      |
| STT VALUE APU       |       nan |                    |
| STT LIMIT dGPU      |       nan | dgpu-skin-temp     |
| STT VALUE dGPU      |       nan |                    |
| CCLK Boost SETPOINT |       nan | power-saving /     |
| CCLK BUSY VALUE     |       nan | max-performance    |
```

This only worked after I added the kernel boot parameter `iomem=relaxed`. `DEVMEM` related kernel configuration parameters for the kernel I use as follows:
```
CONFIG_DEVMEM=y
CONFIG_ARCH_HAS_DEVMEM_IS_ALLOWED=y
CONFIG_STRICT_DEVMEM=y
CONFIG_IO_STRICT_DEVMEM=y
```

Using the `ryzen_smu` DKMS module did **not** seem necessary.

Hope this helps anyone :)

moson-mo, 2023-12-07:
@patrickschur How did you get to the table size of `0xAA0`?
https://github.com/FlyGoat/RyzenAdj/commit/052d88446cf6996f3b0f7d1e78d7e1a9f486f407#diff-d1cde62eb4009283d0941836eb0a27e6b5cefcb14721f2cde3af994e9d493b99R175

Or is that just a good guess based on looking at the data? :)

patrickschur, 2023-12-07:
@moson-mo I don't know the exact table size. But I can tell you the size is at least `0xAA0` based on the table dumps we got. The number actually comes from my script which was used to dump the table. https://gist.github.com/patrickschur/4645f81601fe6f8ff16c7a90e848d7cb#file-pmtable-py-L63

GreyXor, 2023-12-08:
Thank you, yes, I've tested it and it works well for me.

closed 2023-12-08 by GreyXor