FlyGoat/RyzenAdj #265: Any option to change the CPU governor?

firefoxlover, 2023-09-28:
Hey,

I am wondering if there could or already is an option to change the general CPU governor into a powersave mode.

This would be the most normal way to save energy, and working on all machines

FlyGoat, 2023-09-29:
Hi, since governor is a OS thing this is our of our scope.  This project aims to be a bridge between sw and hw.

Thanks

schyhub, 2024-03-31:
> Hey,
> 
> I am wondering if there could or already is an option to change the general CPU governor into a powersave mode.
> 
> This would be the most normal way to save energy, and working on all machines

Agree with Flygoat, also if Linux, you just need to `echo power | sudo tee /sys/...` at the right path to have them in powersave