FlyGoat/RyzenAdj #266: Power saving and performance options don't work on Windows for Ryzen 7940HS

Neurrone, 2023-10-02:
As per title. I'm on Windows 11 on an Elitebook 845 G10.

Falcosc, 2023-10-03:
That's normal, a result of the discussion how the parameter should be named. It only affects the boost delay, so very minor change in performance. But measurable battery life effect under controlled environment. Impossible to feel the difference (that's why AMD did hide it because it would confuse users if they don't feel the difference)

And it's already applied automatically

https://github.com/FlyGoat/RyzenAdj/wiki/FAQ#how-to-fix-the-boost-delay-issue
https://github.com/FlyGoat/RyzenAdj/wiki/FAQ#can-i-use---max-performance-while-on-cable
https://github.com/FlyGoat/RyzenAdj/wiki/FAQ#does---power-saving-improve-battery-runtime

https://www.reddit.com/r/Amd/comments/mgmxjk/ryzenadj_we_did_discover_a_way_to_remove_boost/

But since the change is so minor and only effecting the first fraction of a second in any benchmark, the latest gen may doesn't have it anymore

ciphray, 2023-10-25:
One area you should be able to see a clear difference of it working is your drive speed
if PSPP is enabled you should see the drive drop down in speed while on battery, if you trigger the Max Performance option it should reach full speed instead in this case