FlyGoat/RyzenAdj #271: Save/revert [default] system state

birdie-github, 2023-11-24:
Would be great to have an option to save to/revert to default changes.

E.g. `ryzenadj --save-state=/tmp/default-state -change1 -change2 -change3`

And then `ryzenadj --restore-state=/tmp/default-state`

I absolutely need it 'cause otherwise on suspend-resume my system starts royally misbehaving due to ryzenadj changes.

Also that would be a nice way to restore whatever experimental options you've enabled.

Thank you.

FlyGoat, 2023-11-24:
This is pretty hard to implement because there are many states we can't read from the hardware, we have no idea what's the default value either.
Also, some of SMU calls appears to be events instead of persistent states.

Falcosc, 2023-11-24:
And even harder on Windows, because you need to get access to memory locations which aren't part of our program, and this is raising red flags on antivirus software or anti cheat software.

The best way to solve this is figuring out which values you want to go back to

`ryzenadj -option1=100 -option2=50 -option3=40`
and then put your original values back in to revert: `ryzenadj -option1=80 -option2=40 -option3=30`

closed 2023-11-24 by birdie-github