FlyGoat/RyzenAdj #272: [Not a bug] Phoenix gets stuck at ~500MHz after resume after using --power-saving

birdie-github, 2023-11-26:
My 7840HS based laptop consistently gets stuck around 500MHz on suspend-resume after running this:

```
ryzenadj --tctl-temp=75 --power-saving --stapm-limit=30000 --fast-limit=30000 --slow-limit=20000
```

I guess out of these options `--power-saving` is wrecking my system but I've no idea how to revert it.

`--max-performance` doesn't sound like its counterpart.

I'm sad because without `--power-saving` my CPU idles around 4W and with this option it's down to 0.8W.

Falcosc, 2023-11-26:
It is, I updated wiki to make it more clear https://github.com/FlyGoat/RyzenAdj/wiki/FAQ#can-i-use---max-performance-while-on-cable

I have seen similar behavior. On small background loads like the resource wasting webex teams, it is very noticeable. This tool is doing so much stupid stuff during idle that AMDs more sensitive boost setting does begin to raise the clock speeds too much on Renoir. Interesting that the idle difference is as high with Phoenix.
On a system with less intrusive efficient chat tools you will not see any difference with or without the delay in clock raise.

But if you get stuck at 500Mhz it sounds like `--power-saving` interface could be slightly incorrect for Phoenix (interface definition was copied from older generations by a brave assumption it would be equal).

Power saving is basically simulating one aspect of the battery profile. Which means if the issue doesn't existing without the option on battery mode, then your battery mode doesn't set it or phoenix does set the bit slightly differently when you switch to battery.

birdie-github, 2023-11-27:
Sadly, regardless whether it's Windows or Linux, switching to battery power doesn't change anything in terms of idle CPU power consumption. Could be a BIOS bug, could be something else.

BTW, I'm starting to believe the suspend-resume CPU bug could manifest itself even without using ryzenadj. I'll update the ticket when I am confident what's going on.

closed 2023-11-27 by birdie-github

Falcosc, 2023-11-27:
> switching to battery power doesn't change anything in terms of idle CPU power consumption. 
It is difficult to proof whether your device is setting the power-save option or not. But if you could proof it, then maybe it was not used because of bugs. But this is just an assumption. We found the interface by monitoring what values got written into registers while switching from cable to battery.