FlyGoat/RyzenAdj #278: Ryzen ADJ wont run

900Ninety, 2023-12-20:
So I followed the guide but get the error when running commands to change performance parameters 
[image]

I have linux Mint 21.2 Mate

900Ninety, 2023-12-20:
`joe@joe-Lenovo-100e-2nd-Gen:~$ git clone https://github.com/FlyGoat/RyzenAdj.git
Cloning into 'RyzenAdj'...
remote: Enumerating objects: 997, done.
remote: Counting objects: 100% (475/475), done.
remote: Compressing objects: 100% (220/220), done.
remote: Total 997 (delta 324), reused 319 (delta 247), pack-reused 522
Receiving objects: 100% (997/997), 319.74 KiB | 1.65 MiB/s, done.
Resolving deltas: 100% (630/630), done.
joe@joe-Lenovo-100e-2nd-Gen:~$ cd RyzenAdj
joe@joe-Lenovo-100e-2nd-Gen:~/RyzenAdj$ rm -r win32
joe@joe-Lenovo-100e-2nd-Gen:~/RyzenAdj$ mkdir build && cd build
joe@joe-Lenovo-100e-2nd-Gen:~/RyzenAdj/build$ cmake -DCMAKE_BUILD_TYPE=Release ..
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type: Release
-- IPO / LTO enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /home/joe/RyzenAdj/build
joe@joe-Lenovo-100e-2nd-Gen:~/RyzenAdj/build$ make
[  8%] Building C object CMakeFiles/ryzenadj.dir/lib/osdep_linux.c.o
[ 16%] Building C object CMakeFiles/ryzenadj.dir/lib/nb_smu_ops.c.o
[ 25%] Building C object CMakeFiles/ryzenadj.dir/lib/api.c.o
[ 33%] Building C object CMakeFiles/ryzenadj.dir/lib/cpuid.c.o
[ 41%] Building C object CMakeFiles/ryzenadj.dir/argparse.c.o
[ 50%] Building C object CMakeFiles/ryzenadj.dir/main.c.o
[ 58%] Linking C executable ryzenadj
[ 58%] Built target ryzenadj
[ 66%] Building C object CMakeFiles/libryzenadj.dir/lib/osdep_linux.c.o
[ 75%] Building C object CMakeFiles/libryzenadj.dir/lib/nb_smu_ops.c.o
[ 83%] Building C object CMakeFiles/libryzenadj.dir/lib/api.c.o
[ 91%] Building C object CMakeFiles/libryzenadj.dir/lib/cpuid.c.o
[100%] Linking C shared library libryzenadj.so
[100%] Built target libryzenadj
joe@joe-Lenovo-100e-2nd-Gen:~/RyzenAdj/build$ if [ -d ~/.local/bin ]; then ln -s ryzenadj ~/.local/bin/ryzenadj && echo "symlinked to ~/.local/bin/ryzenadj"; fi
joe@joe-Lenovo-100e-2nd-Gen:~/RyzenAdj/build$ if [ -d ~/.bin ]; then ln -s ryzenadj ~/.bin/ryzenadj && echo "symlinked to ~/.bin/ryzenadj"; fi
joe@joe-Lenovo-100e-2nd-Gen:~/RyzenAdj/build$ ./ryzenadj --stapm-limit=45000 --fast-limit=45000 --slow-limit=45000 --tctl-temp=90
pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
PCI Bus is not writeable, check secure boot
Unable to get MP1 SMU Obj
Unable to init ryzenadj
joe@joe-Lenovo-100e-2nd-Gen:~/RyzenAdj/build$ `

FlyGoat, 2023-12-21:
You might need to disable secure boot in bios.

900Ninety, 2023-12-21:
> You might need to disable secure boot in bios.

I did try turning it on and off , same error 

When I close terminal and reopen , the command cannot find the RyzenADJ folder for some reason

dtl131, 2023-12-22:
You need root permissions.

Agno94, 2024-01-21:
Please run 
```
sudo cat /sys/kernel/security/lockdown
```
If the result has `integrity` selected then kernel lockdown.

dahabakuk, 2024-02-24:
if secure boot is disabled and u have root permission to access /dev/mem it might be `*_STRICT_DEVMEM=y` in which case u will also need kernel paramter:
`iomem=relaxed`

this took me awhile but i got it working now...

dtl131, 2024-02-25:
I think you can close this. The user said it worked once they ran it as root: https://forums.linuxmint.com/viewtopic.php?p=2412555#p2412555

closed 2024-02-26 by Falcosc