FlyGoat/RyzenAdj #279: Is there a way to totally silence the output?

Kabouik, 2023-12-23:
I am trying to write a script for `ryzenadj` that will change the TDP by +1W, -1W or directly to the desired TDP value supplied. I got something working, but even when I append `2>/dev/null` or `>/dev/null 2>&1` to the `ryzenadj` command, I still get a lot of `SMU_SERVICE` lines. How do I suppress them?

tallpeak, 2023-12-23:
You might find something interesting in my script https://github.com/tallpeak/AHK/blob/main/GTA/AFKMoney/RyzenAdj.ahk

Kabouik, 2023-12-23:
Thanks!

Actually I finally almost succeeded, the `SMU_SERVICE` were a mistake on my end, they were coming from the line where my script reads the current TDP. Now I only have the `success` lines that I couldn't silence yet, but the script is working: 

https://github.com/FlyGoat/RyzenAdj/assets/7107523/414af863-d60e-41ec-af04-23fc081d7627

(I also added `min` and `max` as compatible keywords after the video.)

It's a stupid script and most of it could be done with just some rather long `ryzenadj` commands (except incremental changes), but it's [here](https://git.sr.ht/~mlaparie/tdpadj) if anyone needs it. For me the benefit was incremental changes, so that I can set two global keybindings to increase or decrease TDP in my WM.