FlyGoat/RyzenAdj #280: Ryzen 9 7945HX Support?

steffsl, 2024-01-04:
Hello,

I used version  "dac383e1cd Add experimental Hawk Point support" and this seems to not support the Ryzen 9 7945HX.

When I run (or any ryzenadj command for that matter)
`sudo ./ryzenadj --dump-table`

i get the following error:

`Fam19h: unsupported model 97
Only Ryzen Mobile Series are supported
Unable to init ryzenadj`

Output from `cat /proc/cpuinfo`

_processor       : 0
vendor_id       : AuthenticAMD
cpu family      : 25
model           : 97
model name      : AMD Ryzen 9 7945HX with Radeon Graphics
stepping        : 2
microcode       : 0xa601203
cpu MHz         : 400.000
cache size      : 1024 KB
physical id     : 0
siblings        : 32
core id         : 0
cpu cores       : 16
apicid          : 0
initial apicid  : 0
fpu             : yes
fpu_exception   : yes
cpuid level     : 16
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good amd_lbr_v2 nopl nonstop_tsc cpuid extd_apicid aperfmperf rapl pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 x2apic movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs skinit wdt tce topoext perfctr_core perfctr_nb bpext perfctr_llc mwaitx cpb cat_l3 cdp_l3 hw_pstate ssbd mba perfmon_v2 ibrs ibpb stibp ibrs_enhanced vmmcall fsgsbase bmi1 avx2 smep bmi2 erms invpcid cqm rdt_a avx512f avx512dq rdseed adx smap avx512ifma clflushopt clwb avx512cd sha_ni avx512bw avx512vl xsaveopt xsavec xgetbv1 xsaves cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local avx512_bf16 clzero irperf xsaveerptr rdpru wbnoinvd cppc arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic v_vmsave_vmload vgif x2avic v_spec_ctrl vnmi avx512vbmi umip pku ospke avx512_vbmi2 gfni vaes vpclmulqdq avx512_vnni avx512_bitalg avx512_vpopcntdq rdpid overflow_recov succor smca fsrm flush_l1d
bugs            : sysret_ss_attrs spectre_v1 spectre_v2 spec_store_bypass srso
bogomips        : 4990.66
TLB size        : 3584 4K pages
clflush size    : 64
cache_alignment : 64
address sizes   : 48 bits physical, 48 bits virtual
power management: ts ttp tm hwpstate cpb eff_freq_ro [13] [14]_

Is it planned to support this CPU?
If you need more data, please tell me and i'll do my best to provide you with it.

FlyGoat, 2024-01-04:
Reading settings from CPU is currently impossible as we don’t have PM table fields for Hawk Point family, however performing some settings should work.

steffsl, 2024-01-04:
Thank you very much for answering.

That's a shame that we don't have the required data to support it properly.
Is the 7945HX similar to Hawk Point? Because according to the AMD Specs page ( https://www.amd.com/en/product/13016  ) it's supposed to be "Dragon Range". Though truth be told, i don't really know a lot about the different Generations or Product Families.

>"performing some settings should work"
So you suggest I could try adding this CPU as part of the Hawk Point family and check how much of it is working?

steffsl, 2024-01-04:
Well, i tried setting the Family of this CPU first to Phoenix then to Hawk Point. The Behaviour is the same, it gets Stuck in nb_smu_ops.c line 59 ( function: static int smu_service_test(smu_t smu) ) waiting for this response to change, which never happens: response = smn_reg_read(smu->nb, smu->rep);

steffsl, 2024-01-04:
I am very sorry. I completely overlooked an issue that has been opened a while ago about the same topic: https://github.com/FlyGoat/RyzenAdj/issues/254

Flashwalker, 2024-03-04:
I have the same issue on Ryzen 9 7945HX