FlyGoat/RyzenAdj #284: Problem with compilation, on arch linux via AUR and  from git

MaksWisniewski, 2024-03-02:
While having installed all the necessary dependencies, it is unable to compile.

[  8%] Building C object CMakeFiles/ryzenadj.dir/lib/osdep_linux.c.o
In file included from /home/max/.cache/yay/ryzenadj-git/src/ryzenadj/lib/osdep_linux.c:10:
/home/max/.cache/yay/ryzenadj-git/src/ryzenadj/lib/nb_smu_ops.h:80:10: fatal error: pci/pci.h: No such file or directory
   80 | #include <pci/pci.h>
      |          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ryzenadj.dir/build.make:76: CMakeFiles/ryzenadj.dir/lib/osdep_linux.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:85: CMakeFiles/ryzenadj.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
 -> error making: ryzenadj-git-exit status 4
 -> Failed to install the following packages. Manual intervention is required:
ryzenadj-git - exit status 4

Falcosc, 2024-03-02:
Do you have pci libs installed?

> Please make sure that you have libpci dependency before compiling. On Debian-based distros this is covered by installing pcilib-dev package:
>
> sudo apt install build-essential cmake libpci-dev
>
> On Fedora:
>
> sudo dnf install cmake gcc-c++ pciutils-devel

I don't know in which arch package pci libs are. If you find out, it would be nice to share it.

MaksWisniewski, 2024-03-02:
As i've written in the topic of an issue i use arch . Thus dependencies are installed via packege manager.

I've found out that the python package was breaking making of the project.

closed 2024-03-02 by MaksWisniewski