FlyGoat/RyzenAdj #286: readjustService: unnecessarily multiplies tctl-temp by 0.001

xak2000, 2024-03-15:
`readjustService.ps1` always multiplies target value by 0.001:

https://github.com/FlyGoat/RyzenAdj/blob/a46bf6d1b365411ac1d649a51ddbbcffb9849d26/win32/readjustService.ps1#L198-L205

Probably, this is done because most values are defined in `millis` on `set`, but in `millis * 1000` on `get`.
E.g. `fast-limit` could be set to `46000`, but read operation presumably returns `46`.

But this is not the case for `tctl-temp`. We set it as `80` and we get it as `80`.

The script doesn't take it into account, so it prints a warning on each set of the `tctl_temp`:
```
Warning - tctl_temp adjust result 85 does not match target value 0.085. Value 85 will be used for monitoring
```

Full log for the reference:
```
Test Adjustments
set new monitoring target tctl_temp to 0.08
set tctl_temp to 80
set new monitoring target tctl_temp to 0.085
set tctl_temp to 85
Warning - tctl_temp adjust result 85 does not match target value 0.085. Value 85 will be used for monitoring
Monitor tctl_temp changes and PowerSlider changes and ACLineStatus changes every 30 seconds...
tctl_temp value unexpectedly changed from 85 to 100
set tctl_temp to 85
Warning - tctl_temp adjust result 85 does not match target value 0.085. Value 85 will be used for monitoring
```

This is not a major problem as the script still works fine. Fortunately it reads the value after set and then uses `monitorFieldAdjResult` instead of `monitorFieldAdjTarget`.

The problem is just unnecessary warnings in the logs.

Falcosc, 2024-03-15:
I don't know any use case where temp values get overwritten, for that reason the script only works for milliwatt based values to reduce complexity of the code. Doesn't help if it gets even more complex.
So I don't suggest a patch for this, except you have one which uses less code then the current version :)

Falcosc, 2024-03-15:
One version would be Ok: you could test a version where it checks for both values, 0.085 and 85
and if one of both values matches, it proceeds. That would avoid the complexity of having field types in this logic.

Feel free to experiment with this little change, we can merge it if it works great for some days :)

xak2000, 2024-03-16:
Thanks! I agree about the complexity of the script. I will think how can I make it no more complex, but still take into account the fact that temps do not need to be divied by 1000. Checking for both 0.085 and 85 feeling too hacky and maybe can produce false-positives in some cases, so I do not like this solution.

---

Not sure I understand what do you mean by:
> I don't know any use case where temp values get overwritten

I just configured the script to monitor `tctl_temp` and adjust the temp to 85 celsium degrees.
```
$monitorField = "tctl_temp"
$Script:repeatWaitTimeSeconds = 30
adjust "tctl_temp" 85
```

My use case is to prevent CPU to be too hot while gaming to reduce the fans noise. I noticed, that usually games do not need too much CPU power but are still trying to consume it. I can limit the power, of course, but I found limiting the temp more appropriate here as the full power can be used until temp limit is reached. Also, it is tricky to select such power limit that will not let CPU to be too hot, but at the same time will not limit it too much. Especially when a discrete GPU is involved too, that adds to the hotness of the Laptop.

TLDR: As my main and the only goal is to limit the temp and noise, I decided that limiting the temp directly instead of indirectly through wattage is better/simplier approach.

Falcosc, 2024-03-16:
With "I don't know any use case where temp values get overwritten" I wanted to suggest you may don't need to enable monitoring on tctl_temp. On my PC it did not need to be set multiple times. But maybe you actually need tctl monitoring on your PC.

Feel free to play around with solutions.
I don't consider a value difference by factor 1000 as false positive. Values can only fluctuate by factor 0.1 to 3. Everything above factor 10 can not happen. For that reason I did suggest this hack since there are no false positives for factor 1000.
And on errors, like no value or wrong value was collected, the factor is either empty or anything else, but never exactly 1000 :)

Why do I explain it in detail?: Don't focus on unrealistic use cases, reducing complexity is more important than supporting edge cases, which never happens. And even if one of many million uses will encounter values errors exactly off by factor 1000. I would accept this issue, because nothing bad happens. But having to complex code hurts the millions of use cases which are not effected by the edge case.

xak2000, 2024-03-17:
Ah, I see. Yes, on my Legion 5 Pro with Ryzen 6800H something periodically restores the settings of current profile (Legion has 3 profiles: quiet/balance/performance). So, it is not enough to set `tctl_temp` just one time. It constantly returns to `100` every 5-10 mins.

And 100 C doesn't make sense for most games, as it gives close to zero performance increase, while increases the noise (and, probably, the wearout). Also, when CPU throttle is happening at 100 C, then game freezes as wattage is reduced to the ground for a second. While e.g. on 80 C, this hard throttling is not happening - just a small wattage decrease (soft throttling I would call it?) from time to time to keep temp at roughly 80 C.

Thank you for the response and for the explanation while factor of 1000 is safe enough to ignore almost impossible edge cases. I agree with your explanation!

xak2000, 2024-03-17:
Hmm.. one problem with this solution is that even if it will help to suppress the warning, the log will still say:

```
set new monitoring target tctl_temp to 0.085
set tctl_temp to 85
```

So, basically the logs are lying, saying that monitoring target will be `0.085`. Or, rather, technically it doesn't lie, as there is no unit of measure displayed here :), but the log line is still somewhat confusing as ryzenadj doesn't return the value as `0.085`. It returns the value as `85`.

Maybe the better solution would be to just directly check if the field is `tctl_temp` and to not multiply it by `0.001` in this case.

I.e. change the line 200 of `adjust` function from the first comment from:

```ps1
$newTargetValue = [math]::round($value * 0.001, 3, 0)
```

to:

```ps1
$multiplier = if($fieldName -eq "tctl_temp") { 1 } else { 0.001 }
$newTargetValue = [math]::round($value * $multiplier, 3, 0)
```

With this code the logs looks like this:

```
Test Adjustments
set new monitoring target tctl_temp to 80
set tctl_temp to 80
set new monitoring target tctl_temp to 85
set tctl_temp to 85
Monitor tctl_temp changes and PowerSlider changes and ACLineStatus changes every 30 seconds...
tctl_temp value unexpectedly changed from 85 to 100
set tctl_temp to 85
```

Maybe this conditional value transformation is somewhat strange, but I think the root of the problem lies in the fact that ryzenadj inputs do not match outputs for some fields. For example `set_slow_limit` expects the parameter as `25000` (in milliwatts), but `get_slow_limit` returns `25` (in watts). At the same time `set_tctl_temp` expect the parameter as `85` (degrees Celsius) and returns as `85` (still degrees Celsius). I do not know why the interface is implemented this way. This makes the behavior inconsistent and forces to use workarounds like this.

Probably, the most correct soluton would be to always transform the result of `get_*` function by a "normalization" function, that will be aware of every field that is returned in a different unit of measurement than was used to set it, and will transform the result to the same unit of measurement that was used to `set_*` the value. I'm not sure what fields exactly are returned in different unit of measurement. Probably all, expect `*_temp` fields? Then, the implementation of such normalization function could be very simple. Almost as the code above, but with `apu_skin_temp` and `dgpu_skin_temp` in addition to `tctl_temp`.

Falcosc, 2024-03-17:
This give me this idea:
Multiplier = if value > 2000 then 0.001 else 1

Could you add it to a pull request and test it for some days?

xak2000, 2024-03-17:
I created a PR draft. Tested it with `tctl_temp` and `slow_limit` as `$monitorField`. Works fine for now.

PR #287 (2024-03-19): Improve monitorField target value calculation
The `monitorField` target value calculation now uses heuristic method to convert the target value to the appropriate unit of measurement, that RyzenAdj uses to return value in.

E.g. `fast_limit` will be divided by 1000, as RyzenAdj sets the value in milliwatts (e.g. 25000), while returns the value in watts (e.g. 25). But `tctl_temp` will not be divided by 1000, as RyzenAdj sets the value in degrees Celsius (e.g. 85) and returns the value also in degrees Celsius (e.g. 85).

Fixes #286 

@Falcosc I'm not sure why did you select exactly value 2000 fo [... cut]
commit 6dce42ad (2024-03-19): Improve monitorField target value calculation

The `monitorField` target value calculation now uses heuristic method to convert the target value to the appropriate unit of
measurement, that RyzenAdj uses to return value in.

E.g. `fast_limit` will be divided by 1000, as RyzenAdj sets the value in milliwatts (e.g. 25000), while returns the value in watts (e.g. 25). But `tctl_temp` will not be divided by 1000, as RyzenAdj sets the value in degrees Celsius (e.g. 85) and returns the value also in degrees Celsius (e.g. 85).

Fixes #286
closed 2024-03-19 by Falcosc