FlyGoat/RyzenAdj #290: Doesnt Work With secure boot

MARTINrichard05, 2024-04-09:
Hi
I know it kinda sounds dumb but, with secure boot enabled and kernel in lockdown mode, ryzenadj doesnt work. 
Will there be a chance that it will get integrated in the mainline kernel ?

FlyGoat, 2024-04-10:
I think zero chance here. 
You may check https://github.com/FlyGoat/RyzenAdj/issues/231 for other workarounds.

MARTINrichard05, 2024-04-26:
i see this, but it is still a security flaw, did amd released something that can allow tweaking the cpu on linux?
some kind of exposing the management engine with a "socket", and make possible editing cpu values via /sys/class configs ?

FlyGoat, 2024-04-26:
@MARTINrichard05 As a kernel maintainer I would say zero chance for mainline kernel to accept such thing. All power management stuff are meant to be handled by ACPI and AMD PMF. We are basically bypassing those standard interfaces in RyzenAdj. Building our own out of tree kernel module is possible, but you still have to get the module signed to load on lockdown kernel.

Honestly I don’t understand what’s the point here. Using RyzenAdj means you are a pro user who understand what you are doing, then for sure you can manage your system integrity.

closed 2024-04-29 by FlyGoat