FlyGoat/RyzenAdj #291: Allow disabling IPO/LTO in CMakeLists

ShadowApex, 2024-04-10:
I maintain [Rust bindings](https://gitlab.com/shadowapex/libryzenadj-rs) to libryzenadj and encountered linking issues in Rust when `CMAKE_INTERPROCEDURAL_OPTIMIZATION` is set to `TRUE`. Could we update the CMakeLists to include a flag to disable IPO/LTO even if it is supported?

ShadowApex, 2024-04-11:
In the linking step at the end of the Rust build, it fails to find the symbols.

Small snippet from attached log:

```
...
  = note: /usr/bin/ld: /home/william/Projects/PowerStation/target/debug/deps/liblibryzenadj-ec6b339ba685ce56.rlib(libryzenadj-ec6b339ba685ce56.1pc6pnmuzuak76qe.rcgu.o): in function `libryzenadj::RyzenAdj::new':
          /home/william/Projects/libryzenadj-rs/src/lib.rs:89: undefined reference to `init_ryzenadj'
          /usr/bin/ld: /home/william/Projects/libryzenadj-rs/src/lib.rs:94: undefined reference to `init_table'
          /usr/bin/ld: /home/william/Projects/PowerStation/target/debug/deps/liblibryzenadj-ec6b339ba685ce56.rlib(libryzenadj-ec6b339ba685ce56.1pc6pnmuzuak76qe.rcgu.o): in function `libryzenadj::RyzenAdj::get_fast_limit':
          /home/william/Projects/libryzenadj-rs/src/lib.rs:220: undefined reference to `get_fast_limit'
          /usr/bin/ld: /home/william/Projects/PowerStation/target/debug/deps/liblibryzenadj-ec6b339ba685ce56.rlib(libryzenadj-ec6b339ba685ce56.1pc6pnmuzuak76qe.rcgu.o): in function `libryzenadj::RyzenAdj::get_stapm_limit':
          /home/william/Projects/libryzenadj-rs/src/lib.rs:315: undefined reference to `get_stapm_limit'
...
```

[lto-compile-failure.log](https://github.com/FlyGoat/RyzenAdj/files/14951335/lto-compile-failure.log)

Disabling LTO fixes the issue.

Falcosc, 2024-04-11:
Do you rebuild ryzenadj in your setup? Looks like it.
It feels like your build configuration doesn't know that these methods are exposed interface endpoints, because your build did maybe inline these, if disabling it did fix it.

But yes, I don't think INTERPROCEDURAL_OPTIMIZATION makes any difference at our already short call stack, so it is a quick and easy fix on our side. Just wanted to remind you of a potential hidden secondary issue in your build setup.

Maybe we wait some days and let this question open before we do it in case somebody knows it better than us :)