FlyGoat/RyzenAdj #295: --power-saving increases power draw by ~5W - Ryzen 5825U

fanoush, 2024-05-25:
ryzenadj works quite well, however `--power-saving` is quite puzzling.
My system idles at 12-13.5W, here is default output of `--info`
```
C:\>ryzenadj.exe --info
CPU Family: Cezanne
SMU BIOS Interface Version: 19
Version: v0.15.0
PM Table Version: 400005
|        Name         |   Value   |     Parameter      |
|---------------------|-----------|--------------------|
| STAPM LIMIT         |    15.000 | stapm-limit        |
| STAPM VALUE         |     3.522 |                    |
| PPT LIMIT FAST      |    30.000 | fast-limit         |
| PPT VALUE FAST      |    11.058 |                    |
| PPT LIMIT SLOW      |    25.000 | slow-limit         |
| PPT VALUE SLOW      |     3.603 |                    |
| StapmTimeConst      |   275.000 | stapm-time         |
| SlowPPTTimeConst    |     4.000 | slow-time          |
| PPT LIMIT APU       |    25.000 | apu-slow-limit     |
| PPT VALUE APU       |     3.603 |                    |
| TDC LIMIT VDD       |    33.000 | vrm-current        |
| TDC VALUE VDD       |     0.350 |                    |
| TDC LIMIT SOC       |    13.000 | vrmsoc-current     |
| TDC VALUE SOC       |     1.740 |                    |
| EDC LIMIT VDD       |    90.000 | vrmmax-current     |
| EDC VALUE VDD       |    90.000 |                    |
| EDC LIMIT SOC       |    17.000 | vrmsocmax-current  |
| EDC VALUE SOC       |     0.000 |                    |
| THM LIMIT CORE      |    90.000 | tctl-temp          |
| THM VALUE CORE      |    42.817 |                    |
| STT LIMIT APU       |     0.000 | apu-skin-temp      |
| STT VALUE APU       |     0.000 |                    |
| STT LIMIT dGPU      |     0.000 | dgpu-skin-temp     |
| STT VALUE dGPU      |     0.000 |                    |
| CCLK Boost SETPOINT |    50.000 | power-saving /     |
| CCLK BUSY VALUE     |    72.327 | max-performance    |

```
When I run `ryzenadj.exe  --power-saving` the power consumption at idle goes up to 16-18.5W
```
CPU Family: Cezanne
SMU BIOS Interface Version: 19
Version: v0.15.0
PM Table Version: 400005
|        Name         |   Value   |     Parameter      |
|---------------------|-----------|--------------------|
| STAPM LIMIT         |    15.000 | stapm-limit        |
| STAPM VALUE         |     5.019 |                    |
| PPT LIMIT FAST      |    30.000 | fast-limit         |
| PPT VALUE FAST      |     9.762 |                    |
| PPT LIMIT SLOW      |    25.000 | slow-limit         |
| PPT VALUE SLOW      |     4.834 |                    |
| StapmTimeConst      |   275.000 | stapm-time         |
| SlowPPTTimeConst    |     4.000 | slow-time          |
| PPT LIMIT APU       |    25.000 | apu-slow-limit     |
| PPT VALUE APU       |     4.834 |                    |
| TDC LIMIT VDD       |    33.000 | vrm-current        |
| TDC VALUE VDD       |     2.624 |                    |
| TDC LIMIT SOC       |    13.000 | vrmsoc-current     |
| TDC VALUE SOC       |     1.696 |                    |
| EDC LIMIT VDD       |    90.000 | vrmmax-current     |
| EDC VALUE VDD       |    75.223 |                    |
| EDC LIMIT SOC       |    17.000 | vrmsocmax-current  |
| EDC VALUE SOC       |     0.000 |                    |
| THM LIMIT CORE      |    90.000 | tctl-temp          |
| THM VALUE CORE      |    43.786 |                    |
| STT LIMIT APU       |     0.000 | apu-skin-temp      |
| STT VALUE APU       |     0.000 |                    |
| STT LIMIT dGPU      |     0.000 | dgpu-skin-temp     |
| STT VALUE dGPU      |     0.000 |                    |
| CCLK Boost SETPOINT |    95.000 | power-saving /     |
| CCLK BUSY VALUE     |    40.926 | max-performance    |
```
and the only change seems to be 
```
| CCLK Boost SETPOINT |    95.000 | power-saving /     |

```

When I run `ryzenadj.exe  --max-performance` it goes back to 12-13.5W and

```
| CCLK Boost SETPOINT |    50.000 | power-saving /     |

```

So ` --max-performance` is actually better than `--power-saving` :-)
The measuring is done including power adapter  smart AC power socket. Reported minimal values are like this

```
Voltage | 231.1 | V
Current | 0.096 | A
Active Power | 12.0 | W
Apparent Power | 22.2 | VA
Reactive Power | 18.7 | var
Power Factor | 0.53 |  
```
vs e.g.
```
Voltage | 230.3 | V
Current | 0.128 | A
Active Power | 16.5 | W
Apparent Power | 29.5 | VA
Reactive Power | 24.5 | var
Power Factor | 0.55
```
power consumption is 'Active Power'.

The system is Topton Fanless AMD Mini PC Ryzen7 5825U Computer
available e.g. here https://www.aliexpress.com/item/1005006141287407.html

Other options work quite well, I can tune performance down e.g. by 
`ryzenadj.exe --tctl-temp=85  --apu-skin-temp=45 --stapm-limit=10000  --fast-limit=20000 --stapm-time=64 --slow-limit=15000 --slow-time=128` and I see CPU clock and temperature going down when running games so it is definitely very useful tool with this fanless box. Without ryzenadj the case becomes too hot to touch when running games and the performance is still good enough when tuned down.

Falcosc, 2024-05-25:
--max-performance and --power-saving just sets a hidden flag which is activated automatically when you switch from battery to AC or from AC to battery

My idea of naming it AC-Mode and Battery-Mode was overruled. But we don't know what this flag actually is, we just know that it's power-saving when on battery and max-performance on AC.

We don't know if the values for the Boost are stored somewhere. It looks like the flag just triggers the value change.

fanoush, 2024-05-26:
ok, checked the source and looks like you are just sending `_do_adjust(0x12);` or `_do_adjust(0x11);`
https://github.com/FlyGoat/RyzenAdj/blob/ad7f6205d14a55581a0a38c83cabc95745779c8b/lib/api.c#L1111
so it is just what the BIOS does with it. If it helps below  is `--power-saving` with `--dump-table` , not sure if anything there explains why power draw goes up. Anyway, you can close the issue if you feel so as there is no fix possible (except maybe renaming those options if you get more reports like that).

Thanks.

```
ryzenadj.exe --dump-table --power-saving
Sucessfully enable power_saving
PM Table Dump of Version: 400005
| Offset |    Data    |   Value   | After Adjust |
|--------|------------|-----------|--------------|
| 0x0000 | 0x41700000 |    15.000 |       15.000 |
| 0x0004 | 0x406BEEDB |     3.686 |        3.535 |
| 0x0008 | 0x41F00000 |    30.000 |       30.000 |
| 0x000C | 0x4068573C |     3.630 |       10.645 |
| 0x0010 | 0x41C80000 |    25.000 |       25.000 |
| 0x0014 | 0x40685CE9 |     3.631 |        3.190 |
| 0x0018 | 0x41C80000 |    25.000 |       25.000 |
| 0x001C | 0x40685CE9 |     3.631 |        3.190 |
| 0x0020 | 0x42040000 |    33.000 |       33.000 |
| 0x0024 | 0x3F0A255A |     0.540 |        0.192 |
| 0x0028 | 0x41500000 |    13.000 |       13.000 |
| 0x002C | 0x3FA7CAB7 |     1.311 |        1.286 |
| 0x0030 | 0x42B40000 |    90.000 |       90.000 |
| 0x0034 | 0x41EE7177 |    29.805 |       65.650 |
| 0x0038 | 0x41880000 |    17.000 |       17.000 |
| 0x003C | 0x00000000 |     0.000 |        0.000 |
| 0x0040 | 0x42B40083 |    90.001 |       90.001 |
| 0x0044 | 0x4204C7BE |    33.195 |       39.159 |
| 0x0048 | 0x42B40083 |    90.001 |       90.001 |
| 0x004C | 0x41FE941F |    31.822 |       32.659 |
| 0x0050 | 0x42B40083 |    90.001 |       90.001 |
| 0x0054 | 0x4201067F |    32.256 |       32.983 |
| 0x0058 | 0x00000000 |     0.000 |        0.000 |
| 0x005C | 0x00000000 |     0.000 |        0.000 |
| 0x0060 | 0x00000000 |     0.000 |        0.000 |
| 0x0064 | 0x00000000 |     0.000 |        0.000 |
| 0x0068 | 0x450999E5 |  2201.618 |     2199.959 |
| 0x006C | 0x41B3668C |    22.425 |        0.492 |
| 0x0070 | 0x3FBB3333 |     1.462 |        1.462 |
| 0x0074 | 0x3F609016 |     0.877 |        1.135 |
| 0x0078 | 0x41900000 |    18.000 |       18.000 |
| 0x007C | 0x4206231A |    33.534 |        0.000 |
| 0x0080 | 0x00000000 |     0.000 |        0.000 |
| 0x0084 | 0x42C80000 |   100.000 |      100.000 |
| 0x0088 | 0x3F290E80 |     0.660 |        4.032 |
| 0x008C | 0x3F813514 |     1.009 |        1.167 |
| 0x0090 | 0x3FA94115 |     1.322 |        1.500 |
| 0x0094 | 0x3F2248D1 |     0.634 |        0.634 |
| 0x0098 | 0x406851B5 |     3.630 |        7.333 |
| 0x009C | 0x40910C1A |     4.533 |        4.550 |
| 0x00A0 | 0x40911C0E |     4.535 |        4.550 |
| 0x00A4 | 0x40911BDD |     4.535 |        4.550 |
| 0x00A8 | 0x40911C0E |     4.535 |        4.550 |
| 0x00AC | 0x40911C0E |     4.535 |        4.550 |
| 0x00B0 | 0x40911C0E |     4.535 |        4.550 |
| 0x00B4 | 0x40911C0E |     4.535 |        4.550 |
| 0x00B8 | 0x40911C0E |     4.535 |        4.550 |
| 0x00BC | 0x40911C0E |     4.535 |        4.550 |
| 0x00C0 | 0x40910C4B |     4.533 |        4.550 |
| 0x00C4 | 0x40911C0E |     4.535 |        4.550 |
| 0x00C8 | 0x3FFF2204 |     1.993 |        2.000 |
| 0x00CC | 0x40000000 |     2.000 |        2.000 |
| 0x00D0 | 0x3FFF949E |     1.997 |        2.000 |
| 0x00D4 | 0x40000000 |     2.000 |        2.000 |
| 0x00D8 | 0x40000000 |     2.000 |        2.000 |
| 0x00DC | 0x40000000 |     2.000 |        2.000 |
| 0x00E0 | 0x40000000 |     2.000 |        2.000 |
| 0x00E4 | 0x40000000 |     2.000 |        2.000 |
[... 1115 lines cut ...]
| 0x08E8 | 0x3DE6FD19 |     0.113 |        0.114 |
| 0x08EC | 0x00000000 |     0.000 |        0.000 |
| 0x08F0 | 0x42550000 |    53.250 |       42.750 |
| 0x08F4 | 0x3F966666 |     1.175 |        1.162 |
| 0x08F8 | 0x3F0EC015 |     0.558 |        1.629 |
| 0x08FC | 0x3F636AC2 |     0.888 |        1.079 |
| 0x0900 | 0x3BCCF680 |     0.006 |        0.006 |
| 0x0904 | 0x4232D8C1 |    44.712 |       22.660 |
| 0x0908 | 0x00000000 |     0.000 |        0.000 |
| 0x090C | 0x00000000 |     0.000 |        0.000 |
| 0x0910 | 0x00000000 |     0.000 |        0.000 |
| 0x0914 | 0x3F800000 |     1.000 |        1.000 |
| 0x0918 | 0x43898000 |   275.000 |      275.000 |
| 0x091C | 0x40800000 |     4.000 |        4.000 |
| 0x0920 | 0x43C80000 |   400.000 |      400.000 |
| 0x0924 | 0x42C80000 |   100.000 |      100.000 |
| 0x0928 | 0x43C80000 |   400.000 |      400.000 |
| 0x092C | 0x43480000 |   200.000 |      200.000 |
| 0x0930 | 0x43480000 |   200.000 |      200.000 |
| 0x0934 | 0x44160000 |   600.000 |      600.000 |
| 0x0938 | 0x42C80000 |   100.000 |      100.000 |
| 0x093C | 0x42BBD414 |    93.914 |       96.055 |
| 0x0940 | 0x46AFE400 | 22514.000 |    22514.000 |

```

Falcosc, 2024-05-26:
Unfortunately, renaming would break the API (make it incompatible)