FlyGoat/RyzenAdj #301: Question // cannot use Curve Optimizer on 7840HS

Verycuteabbey, 2024-06-16:
my laptop is Lenovo XiaoXinPro 14 APH8

i looked Issues that Phoenix has support added at Dec 7, 2023

but why i cannot use `--set-coall` and `--set-cogfx` for my 7840HS, at Windows 10 i can use it by UXTU

[image]

ryzenadj info

[image]

Panjaksli, 2024-07-26:
Thank AMD for that. It completely locked down anything other than highest end CPU's starting from 5000 series. This only works in x900 models (or the new bad naming equivalent 7xxx).

plastininikolay, 2026-09-24:
Another Phoenix data point: a Ryzen 9 7940HS (the top Phoenix model) in an ASUS ROG Flow X13 GV302XV (BIOS 319), Linux with ryzen_smu. So on this laptop it is not only lower models that are affected.

`--set-coall=-15` reports "Successfully set", and so do the 20-bit form `1048561` and writing the RSMU command `0x07` (SetAllDldoPsmMargin) directly. None of them change anything. I checked by pinning a busy loop to one core and reading that core's voltage from the PM table (version `0x4C0008`, voltage at `0x814`). At the same clock (4.51 GHz) it is 1.073–1.074 V whether CO is 0 or -15. A working -15 should lower it noticeably. Reading CO back with RSMU `0xD5` (GetDldoPsmMargin) does not work on this CPU either: it returns the argument unchanged, so it always looks like 0.

So "Successfully set" only means the SMU accepted the message, not that CO is active. My guess is that the OEM firmware locks it on this model. If you want to check your own laptop, compare core voltage at the same clock before and after setting CO. #417 adds per-core voltage and clock readout for table `0x4C0008` to libryzenadj.

(I did this with AI assistance and ran the measurements myself.)