FlyGoat/RyzenAdj #303: Cpu puts itself on a 0.5 to 0.9GHZ clock

LOL1862, 2024-06-22:
Rx 6800m 
Ryzen:5900Hx 
everytime i install it it puts itself on such a low clock restarted after installed still the same problem windows powerplan fine cpu boost enabled idk where the problem is

kimhuilim, 2024-10-03:
Your VRM's are overheating. Zen3 also is common to that problem. The STAPM value and so are causing it.

kimhuilim, 2024-10-16:
Update here : I kinda discovered where this problems are getting triggered. I have attempted to tweak everything with SmokelessUMAF in HP's BIOS with no avail. Putting custom values towards the STAPM and STT settings. It still occurs until I decided to just disable STT (Skin Temperature Tracking) altogether. Now the problem's gone. I think it backfires to HP's implementation considering the device I use is made up of plastic housing while the temperature sensors are located somewhere unventilated or cannot conduct heat. HWInfo64 logs that the extreme throttling to 2W only happens when CPU skin temp reaches 50°C. Even if the cores and GPU itself is only limited and doesn't exceed above 80°C (Prochot TJMax is 100°C). Disabling STT using UMAF helped me remove the extreme power throttling but for sure will cause some bad things to happen as it is beyond HP's configured and I guess tested limits for CPU skin temp. But as long I guess that I don't overstress the laptop above 80°C, I should be safe.

Sources that also helped me is lots of threads from Legion Go reddit group when Lenovo implemented the control of STAPM and STT. Don't hesitate to correct me if I'm wrong.