FlyGoat/RyzenAdj #309: Support for Ryzen AI 9 HX 365/370

PisonJay, 2024-08-19:
Hi, as the Ryzen AI 9 processors released, can you add support for them?

Mine is an ASUS Vivobook S 16 with HX 365,
`CPU0: AMD Ryzen AI 9 365 w/ Radeon 880M (family: 0x1a, model: 0x24, stepping: 0x0)`

The new Zen 5 processors seem to have family id 0x1a.

boulate, 2024-08-20:
Same here for the Ryzen AI 9 HX 370 :

lscpu
Architecture :                              x86_64
  Mode(s) opératoire(s) des processeurs :   32-bit, 64-bit
  Tailles des adresses:                     48 bits physical, 48 bits virtual
  Boutisme :                                Little Endian
Processeur(s) :                             24
  Liste de processeur(s) en ligne :         0-23
Identifiant constructeur :                  AuthenticAMD
  Identifiant constructeur du BIOS :        Advanced Micro Devices, Inc.
  Nom de modèle :                           AMD Ryzen AI 9 HX 370 w/ Radeon 890M
    Nom de modèle BIOS :                    AMD Ryzen AI 9 HX 370 w/ Radeon 890M            Unknown CPU @ 2.0GHz
    Famille de processeur BIOS :            107
    Famille de processeur :                 26
    Modèle :                                36
    Thread(s) par cœur :                    2
    Cœur(s) par socket :                    12
    Socket(s) :                             1
    Révision :                              0
    multiplication des MHz du/des CPU(s) :  20%
    Vitesse maximale du processeur en MHz : 4367,0000
    Vitesse minimale du processeur en MHz : 599,0000
    BogoMIPS :                              3994,85
    Drapeaux :                              fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good amd_l
                                            br_v2 nopl xtopology nonstop_tsc cpuid extd_apicid aperfmperf rapl pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_lega
                                            cy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs skinit wdt tce topoext perfctr_core perfctr_nb bpext perfctr_llc mwaitx cpb cat_l3 cdp_l3 hw_pstate ss
                                            bd mba perfmon_v2 ibrs ibpb stibp ibrs_enhanced vmmcall fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid cqm rdt_a avx512f avx512dq rdseed adx smap avx512ifma clflushopt
                                             clwb avx512cd sha_ni avx512bw avx512vl xsaveopt xsavec xgetbv1 xsaves cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local user_shstk avx_vnni avx512_bf16 clzero irperf xsavee
                                            rptr rdpru wbnoinvd cppc arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic v_vmsave_vmload vgif x2avic v_spec_ctrl
                                             vnmi avx512vbmi umip pku ospke avx512_vbmi2 gfni vaes vpclmulqdq avx512_vnni avx512_bitalg avx512_vpopcntdq rdpid bus_lock_detect movdiri movdir64b overflow_recov succor sm
                                            ca fsrm avx512_vp2intersect flush_l1d amd_lbr_pmc_freeze
Fonctionnalités de virtualisation :         
  Virtualisation :                          AMD-V
Caches (somme de toutes) :                  
  L1d :                                     576 KiB (12 instances)
  L1i :                                     384 KiB (12 instances)
  L2 :                                      12 MiB (12 instances)
  L3 :                                      24 MiB (2 instances)
NUMA :                                      
  Nœud(s) NUMA :                            1
  Nœud NUMA 0 de processeur(s) :            0-23
Vulnérabilités :                            
  Gather data sampling :                    Not affected
  Itlb multihit :                           Not affected
  L1tf :                                    Not affected
  Mds :                                     Not affected
  Meltdown :                                Not affected
  Mmio stale data :                         Not affected
  Reg file data sampling :                  Not affected
  Retbleed :                                Not affected
  Spec rstack overflow :                    Not affected
  Spec store bypass :                       Mitigation; Speculative Store Bypass disabled via prctl
  Spectre v1 :                              Mitigation; usercopy/swapgs barriers and __user pointer sanitization
  Spectre v2 :                              Mitigation; Enhanced / Automatic IBRS; IBPB conditional; STIBP always-on; RSB filling; PBRSB-eIBRS Not affected; BHI Not affected
  Srbds :                                   Not affected
  Tsx async abort :                         Not affected

Thank you.

mzdluo123, 2024-09-04:
https://github.com/FlyGoat/RyzenAdj/pull/310

FlyGoat, 2024-09-05:
Hi all,

I had integrated table part of #310 and adjustment part based on reverse engineering ACPI table to a development branch.

See: https://github.com/FlyGoat/RyzenAdj/tree/stx-point

Please test and report!

FlyGoat, 2024-09-05:
For Windows builds please download ryzenadj-win64-debug from https://github.com/FlyGoat/RyzenAdj/actions/runs/10723443339

FlyGoat, 2024-09-05:
I received off-issue test reports stating that at least PPT adjustment features are working, will merge test changes into master.

FlyGoat, 2024-09-05:
Reopen as more testing work and PMTable work is still needed

PisonJay, 2024-10-18:
https://www.amd.com/en/developer/uprof.html

NEW, AMD uProf 5.0 is now available (October 10, 2024)
Maybe we can try hacking PM tables now

tova-cup, 2024-12-07:
I can report that on the Ryzen AI 7 PRO 360 only the PPT values are identified in the table

```
CPU Family: Strix Point
SMU BIOS Interface Version: 20
Version: v0.16.0 
PM Table Version: 5d0008
|        Name         |   Value   |     Parameter      |
|---------------------|-----------|--------------------|
| STAPM LIMIT         |     5.000 | stapm-limit        |
| STAPM VALUE         |     2.509 |                    |
| PPT LIMIT FAST      |     5.000 | fast-limit         |
| PPT VALUE FAST      |     3.112 |                    |
| PPT LIMIT SLOW      |     5.000 | slow-limit         |
| PPT VALUE SLOW      |     2.705 |                    |
| StapmTimeConst      |       nan | stapm-time         |
| SlowPPTTimeConst    |       nan | slow-time          |
| PPT LIMIT APU       |       nan | apu-slow-limit     |
| PPT VALUE APU       |       nan |                    |
| TDC LIMIT VDD       |       nan | vrm-current        |
| TDC VALUE VDD       |       nan |                    |
| TDC LIMIT SOC       |       nan | vrmsoc-current     |
| TDC VALUE SOC       |       nan |                    |
| EDC LIMIT VDD       |       nan | vrmmax-current     |
| EDC VALUE VDD       |       nan |                    |
| EDC LIMIT SOC       |       nan | vrmsocmax-current  |
| EDC VALUE SOC       |       nan |                    |
| THM LIMIT CORE      |       nan | tctl-temp          |
| THM VALUE CORE      |       nan |                    |
| STT LIMIT APU       |       nan | apu-skin-temp      |
| STT VALUE APU       |       nan |                    |
| STT LIMIT dGPU      |       nan | dgpu-skin-temp     |
| STT VALUE dGPU      |       nan |                    |
| CCLK Boost SETPOINT |       nan | power-saving /     |
| CCLK BUSY VALUE     |       nan | max-performance    |

```

Also find attached the full table dump.
[ryzen-AI7-dump.txt](https://github.com/user-attachments/files/18046497/ryzen-AI7-dump.txt)

AngelouDi, 2025-05-01:
I have an Asus Vivobook S15 w/ HX 370, uProf is working fine, is there a way to help in improving the compatibility with ryzenadj?

GreyXor, 2025-05-09:
I would be happy to help for Strix Point AI 9 HX 370 compatiblity
Framework 13

```bash
CPU Family: Strix Point
SMU BIOS Interface Version: 21
Version: v0.16.0 
PM Table Version: 5d0009
|        Name         |   Value   |     Parameter      |
|---------------------|-----------|--------------------|
| STAPM LIMIT         |    46.000 | stapm-limit        |
| STAPM VALUE         |     4.167 |                    |
| PPT LIMIT FAST      |    46.000 | fast-limit         |
| PPT VALUE FAST      |     5.618 |                    |
| PPT LIMIT SLOW      |    35.000 | slow-limit         |
| PPT VALUE SLOW      |     4.933 |                    |
| StapmTimeConst      |       nan | stapm-time         |
| SlowPPTTimeConst    |       nan | slow-time          |
| PPT LIMIT APU       |       nan | apu-slow-limit     |
| PPT VALUE APU       |       nan |                    |
| TDC LIMIT VDD       |       nan | vrm-current        |
| TDC VALUE VDD       |       nan |                    |
| TDC LIMIT SOC       |       nan | vrmsoc-current     |
| TDC VALUE SOC       |       nan |                    |
| EDC LIMIT VDD       |       nan | vrmmax-current     |
| EDC VALUE VDD       |       nan |                    |
| EDC LIMIT SOC       |       nan | vrmsocmax-current  |
| EDC VALUE SOC       |       nan |                    |
| THM LIMIT CORE      |       nan | tctl-temp          |
| THM VALUE CORE      |       nan |                    |
| STT LIMIT APU       |       nan | apu-skin-temp      |
| STT VALUE APU       |       nan |                    |
| STT LIMIT dGPU      |       nan | dgpu-skin-temp     |
| STT VALUE dGPU      |       nan |                    |
| CCLK Boost SETPOINT |       nan | power-saving /     |
| CCLK BUSY VALUE     |       nan | max-performance    |
```

vazh2100, 2025-08-16:
from last commit:
sudo ryzenadj -i
detected compatible ryzen_smu kernel module
Unable to get RSMU Obj
Unable to init ryzenadj

atoppi, 2025-09-30:
Lenovo Thinkpad with HX 370 PRO.
I can report that ryzenadj is only able to parse STAPM and PPT FAST/SLOW limits. In particular I'd be interested in reading and setting the apu skin temperature.

zakyhassani, 2025-10-31:
Confirmed on GPD Win Max 2 2025 Ryzen AI HX 370, same behaviour.

PR #311 (2024-09-05): Initial Strix Point support
Partially based on #310.

MP1 stuff reversed from ACPI table dump.

Closes: #309